adafruit/Raspberry-Pi-Installer-Scripts #1: Running the PiTFT fbcp script with normal/0 results in portrait TFT

dragon788, 2017-04-10:
Background:
I had an installation of Raspbian that I upgraded to the latest, when it switched to the Pixel desktop, the PiTFT 3.5" was just showing a white/gray screen.

Procedure:
I ran the script and selected the options below.

Preconfigured scenario or manual (4)
4 (manual)
Device selection
4 (3.5" TFT)  (specifically gen1) 
HDMI rotation (landscape/portrait 90/ landscape 180/ portrait 270)
1 (landscape)
TFT Rotation (MADCTL) 0 / 90 / 180 / 270
1 (0 degrees rotation to match HDMI)

After selecting and confirming these options and rebooting the TFT came up in portrait mode which didn't match my expectations from my selections.

dragon788, 2017-04-10:
I ended up actually having to select option 4 for the TFT rotation (270 degrees) because otherwise the touchscreen calibration was inverted.

ladyada, 2018-04-09:
script has been rewritten!

closed 2018-04-09 by ladyada