adafruit/Raspberry-Pi-Installer-Scripts #6: Watchdog name changed

JacoFourie, 2017-12-04:
The watchdog module name changed from bcm2835_wdog to bcm2835_wdt.

chrisbartley, 2019-02-21:
Thanks!  More detail, for others who might run across the same problem as I did.

I applied the script, but then got this error on boot:

```
[FAILED] Failed to start Load Kernel Modules
See 'systemctl status systemd-modules-load.service' for details.
```

So I ran that command to see what failed:

```
$ sudo systemctl status systemd-modules-load.service
● systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-02-20 18:35:20 EST; 9s ago
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
  Process: 1161 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=1/FAILURE)
 Main PID: 1161 (code=exited, status=1/FAILURE)

Feb 20 18:35:20 bartley-pi-zero systemd[1]: Starting Load Kernel Modules...
Feb 20 18:35:20 bartley-pi-zero systemd-modules-load[1161]: Failed to find module 'bcm2835_wdog'
Feb 20 18:35:20 bartley-pi-zero systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Feb 20 18:35:20 bartley-pi-zero systemd[1]: Failed to start Load Kernel Modules.
Feb 20 18:35:20 bartley-pi-zero systemd[1]: systemd-modules-load.service: Unit entered failed state.
Feb 20 18:35:20 bartley-pi-zero systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
```

Seeing the "Failed to find module 'bcm2835_wdog'", I searched issues here for `bcm2835_wdog`, found this one, and then looked in the script to see how and where it's used.  The script appends `bcm2835_wdog` to the last line of `/etc/modules`.  So I just remounted `/` as read-write (`sudo mount -o remount,rw /`) and then edited `/etc/modules` to change `bcm2835_wdog` to `bcm2835_wdt`.  Rebooted, and the error was gone.  Then re-ran the status command to verify:

```
$ sudo systemctl status systemd-modules-load.service
● systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static; vendor preset: enabled)
   Active: active (exited) since Thu 2016-11-03 13:16:44 EDT; 2 years 3 months ago
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
  Process: 60 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=0/SUCCESS)
 Main PID: 60 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/systemd-modules-load.service

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
```

ladyada, 2019-02-24:
thanks for the test! if that fixes it could you submit a PR? it would get it fixed for others and we'd appreciate it a ton :)