adafruit/Raspberry-Pi-Installer-Scripts #10: Running the PiTFT script removes the ability to SSH into the Pi.

CJeffyB, 2018-04-06:
When setting up a Pi, I have always made sure that I could SSH into it (and have changed the password).

Tonight I ran the new version of this script for the first time, and at some point in the script, it removes the ability to SSH into the system.  There is no indication that this is being done, and I am not sure what (if any) other unnecessary changes are being silently made in the background.

I have attempted re-adding an empty file named "ssh" on the SD card, but that does not alter the situation at all.

This behavior was not present on the version of the script that was located at:
https://raw.githubusercontent.com/adafruit/Adafruit-PiTFT-Helper/master/adafruit-pitft-helper2.sh
That version of the script worked perfectly fine.

Additionally, this new version is requiring additional sources to be added to the repo list:
`Missing Apt repo, please add deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi firmware to /etc/apt/sources.list.d/raspi.list`
This was not a requirement in prior versions.  (For some reason, it explicitly looks for mirrordirector.raspbian.org, and will not accept archive.raspberrypi.org)

This behavior is on a Pi 3 B Plus, with the [ADAFRUIT PITFT - 320X240 2.8" TFT+TOUCHSCREEN FOR RASPBERRY PI](https://www.adafruit.com/product/1601)

brunoramoslu, 2018-04-08:
I confirm this behavior in my Pi 3B+.

The changes are causing the eth0 interface to get a random mac address and that is breaking the DHCP, so the Pi does not get an IP address rendering it inaccessible through the network.

I was able to isolate the change to the /boot/config.txt file changes (see below), simply reverting this file gets you back to a "normal" state.
However I was not able to identify exactly what is causing the break in functionality or how to fix it keeping the required changes for the display to work.

```
--- /boot/config.txt	2018-04-08 09:00:02.000000000 +0000
+++ /boot/config.txt-bad	2018-04-07 10:45:30.000000000 +0000
@@ -54,3 +54,18 @@
 
 # Enable audio (loads snd_bcm2835)
 dtparam=audio=on
+# --- added by adafruit-pitft-helper Sat  7 Apr 10:45:31 UTC 2018 ---
+[pi0]
+device_tree=bcm2708-rpi-0-w.dtb
+[pi1]
+device_tree=bcm2708-rpi-b-plus.dtb
+[pi2]
+device_tree=bcm2709-rpi-2-b.dtb
+[pi3]
+device_tree=bcm2710-rpi-3-b.dtb
+[all]
+dtparam=spi=on
+dtparam=i2c1=on
+dtparam=i2c_arm=on
+dtoverlay=pitft35-resistive,rotate=90,speed=20000000,fps=20
+# --- end adafruit-pitft-helper Sat  7 Apr 10:45:31 UTC 2018 ---
```
This seems to be related to the way the Device Tree is being loaded.
Adding dtdebug=1 to /boot/config.txt gives more information about the Device Tree loading process but I'm still a bit lost in the debug.

When we use the default /boot/config.txt file we can that the file loaded is bcm2710-rpi-3-b-plus.dtb
```
001757.747: No kernel trailer - assuming DT-capable
001757.786: Trying Device Tree file 'bcm2710-rpi-3-b-plus.dtb'
001757.812: brfs: File read: 4867400 bytes
001764.000: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
001764.025: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x4ac548 size 0x5fb7
```
When using the /boot/config.txt file patched by the script we see that we use a different .dtb file, in this case bcm2710-rpi-3-b.dtb.
We also load a new dtb for the pitft pitft35-resistive.dtbo
```
001864.070: No kernel trailer - assuming DT-capable
001864.100: Trying Device Tree file 'bcm2710-rpi-3-b.dtb'
001864.126: brfs: File read: 4867400 bytes
001870.557: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
001870.580: Loading 'bcm2710-rpi-3-b.dtb' to 0x4ac548 size 0x5eb0
```
```
002089.154: dtdebug: Opened overlay file 'overlays/pitft35-resistive.dtbo'
002090.858: brfs: File read: /mfs/sd/overlays/pitft35-resistive.dtbo
002114.984: Loaded overlay 'pitft35-resistive'
```

**Logs:**

Default /boot/config.txt : [device-tree-information-boot-config-txt-default.txt](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/files/1887599/device-tree-information-boot-config-txt-default.txt)

Adafruit Patched /boot/config.txt : [device-tree-information-boot-config-txt-patched-adafruit.txt](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/files/1887598/device-tree-information-boot-config-txt-patched-adafruit.txt)

brunoramoslu, 2018-04-08:
After some more searching and reading about the /boot/config.txt file configuration and options I was able to understand that the problem is apparently coming from force of the dtb for the [pi3] section.

To make it work on my Pi3 B Plus I had to force the correct dtb file.

```diff
--- /boot/config.txt-bad	2018-04-08 21:49:13.000000000 +0000
+++ /boot/config.txt	2018-04-08 21:23:38.000000000 +0000
@@ -65,7 +65,7 @@
 [pi2]
 device_tree=bcm2709-rpi-2-b.dtb
 [pi3]
-device_tree=bcm2710-rpi-3-b.dtb
+device_tree=bcm2710-rpi-3-b-plus.dtb
 [all]
 dtparam=spi=on
 dtparam=i2c1=on
```

At this stage I was unable to confirm if always loading the -plus version of the dtb file it would still be compatible with the regular version of the Pi3 B.

I don't have a Pi3 B to test unfortunately...

Documentation for config.txt : https://www.raspberrypi.org/documentation/configuration/config-txt/README.md

Device Trees, overlays, and parameters : https://www.raspberrypi.org/documentation/configuration/device-tree.md

ladyada, 2018-04-09:
hiya, i dont think we need to specify the dtb anymore so its been removed. seems to work now and tested with a 3B+!

https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/commit/3f690b22562532043511db41612a413ddb79def6

closed 2018-04-09 by ladyada

brydenrh, 2018-04-15:
I still have the same issue described in the OP. If anyone can walk me through the steps like a 3 year old I would appreciate it. The instructions on the adafruit site for the 2.8" Pi TFT (resistance touchscreen) get me to the last step and then it says the "Missing Apt repo, please add deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi firmware to /etc/apt/sources.list.d/raspi.list"
Which might as well be in Chinese because I don't understand a word of it. I'm new to Pi and linux.
Thanks for any help!

brydenrh, 2018-04-15:
BTW I should add that I'm using a brand new Raspberry Pi 3 B+ and the newest Raspbian version and went through the full update and upgrade steps successfully. Can only see the display via HDMI.

ladyada, 2018-04-15:
@brydenrh please post up in the adafruit tech support forums :)

brydenrh, 2018-04-15:
will do...