adafruit/Raspberry-Pi-Installer-Scripts #13: Feature request: Non-interactive installation option for adafruit-pitft.sh

drew-kun, 2018-04-18:
Hi, I am writing an ansible playbook for my raspberry pi automated setup and would like to have some non-interactive way to install pitft drivers.

Say I can create `pitft.config.sh` with that variables wich the `adafruit-pitft.sh` asks to define interactively, so the script will find find this file(say in the same dir), read those predefined vars and quietly do all that it's supposed to do.

This way the script could painlessly be used in automation procedures.

mamoit, 2018-04-18:
I would even dare to say that all the scripts deserve a non-interactive version.
I also need to put the read only script on an ansible playbook.

drew-kun, 2018-04-18:
@mamoit Exactly! What I have written is how pihole installer .sh resolves this problem. You can just create SetupVars.sh script with predefined vars and run the installer something like: `bash pihole_install.sh --unattended`

ladyada, 2018-04-19:
that would be awesome but we're really really bad at shell scripting and don't forsee being able to write this while maintaining the current script. if someone can submit a PR we'd be very happy to look at it!

mamoit, 2018-04-19:
I think that while we're at it we should support both approaches.
I can't really run it since I don't own one of the supported LCDs, but I'll comment the lines that actually do something and check the logic.
To be continued in @drew-kun 's PR.

mamoit, 2018-04-19:
@drew-kun Thanks for your PR, I think it is a good solution for the problem that you are facing (templating to implement an ansible role).
But I also think that if one needs to run 2 or 3 of these scripts, and they all follow this principle things can get messy quite quickly.

My take on this is that the condition for each step to be (or not be) interactive, should not be whether the flag of unattended is on, but whether the variable is defined, so that we can easily do something as complicated as:
1. Source a file containing most of the configs. There is no need to pass it as argument since it is a niche use and one can do it prior to running the script and we would not need to version it (the scripts would still be self contained);
2. Override some other options with flags passed on to the script, which would set those same variables overriding if one was defined in the config file;
3. And the ones that were left undefined leave them for the user to choose interactively. If no variables were defined either by sourcing a config file or by means of command line switches, then the user would be presented with the full set of queries.

drew-kun, 2018-04-21:
Sorry, I am not sure how to do this :((( That pr was the top of my shell scripting skills lol.

drew-kun, 2018-04-26:
@mamoit, I own the Adafruit capacitive 2.8" TFT display (240x320). If you could do modify the script to meet all requirements that you have mentioned, then I would be happy to test it.

steviethecat, 2018-09-12:
Just call it as:
```
echo -e "1\n2\n" | sudo bash adafruit-pitft.sh
```

Same can be done for the other scripts. E.g. `read-only-fs.sh`.
```
echo -e "Y\nN\nN\nN\nY\nY\n" | sudo bash read-only-fs.sh
```
where the '\n' is a return, and the uppercase Y's and N's are the answers. They can be lowercase, but this makes it a bit more readable.

ladyada, 2018-09-12:
nice! thats a good solution :)

drew-kun, 2018-09-15:
This only works in bash and does not work in ansible - the task just hangs infinitely((

kevinkahn, 2020-07-24:
OK - closed my suggestion to merge here.  The suggestions here about echoing into the script only work if you don't change the meaning of the parameters.  Sometime in the last year you added another screen type and instead of putting it at the end so that the selection numbers for existing screens didn't change you put it in the middle of the list.  Thus any solutions of this sort break.  At the very least, if you are adding things don't break what is already there.  That doesn't take resources - just care.

makermelissa, 2020-10-13:
The script has been rewritten in Python and now is able to be non-interactively installed. See https://learn.adafruit.com/adafruit-2-2-pitft-hat-320-240-primary-display-for-raspberry-pi/easy-install for more information on usage.

closed 2020-10-13 by makermelissa