adafruit/Raspberry-Pi-Installer-Scripts #17: Homogenize script permissions

mamoit, 2018-04-29:
```
-rwxrwxr-x adafruit-pitft.sh
-rwxrwxr-x arcade-bonnet.sh
-rw-rw-r-- i2c.sh
-rw-rw-r-- i2samp.sh
-rwxrwxr-x joy-bonnet.sh
-rw-rw-r-- pi-eyes.sh
-rwxrwxr-x pitft-fbcp.sh
-rwxrwxr-x pi-touch-cam.sh
-rw-rw-r-- README.md
-rw-rw-r-- read-only-fs.sh
-rwxrwxr-x retrogame.sh
-rw-rw-r-- rgb-matrix.sh
-rw-rw-r-- rtc.sh
```
Some scripts are executable, others aren't.
I can solve this, just asking if it is intended for some reason.

ladyada, 2018-05-20:
no reason for it, we write the scripts on Windows anyhow & we expect people to +x on run!

closed 2018-05-20 by ladyada

mamoit, 2018-05-21:
I disagree, since for someone who is new to linux and is following instructions from somewhere and just wants to clone the repo and run a script, not having the right permissions (since they will be most probably be cloning on a rpi) can introduce frustration.
This is not a RFC and more of a 'let me know and I'll do it for you'.

ladyada, 2018-05-21:
okdoke if you can submit a PR we can look at it!

mamoit, 2018-05-21:
There you go.

ladyada, 2018-05-21:
yaaaaay!

PR #26 (2018-05-21): Add execute permissions to all scripts
(Re)closes #17.
Does not add execute permissions to `.bak` files.