adafruit/Raspberry-Pi-Installer-Scripts #24: Rationale behind speed=20000000 on pitft35-resistive?

juj, 2018-05-15:
Hi!

In the pitft setup script, there's

```bash
    if [ "${pitfttype}" == "22" ]; then
        overlay="dtoverlay=pitft22,rotate=${pitftrot},speed=64000000,fps=30"
    fi

    if [ "${pitfttype}" == "28r" ]; then
        overlay="dtoverlay=pitft28-resistive,rotate=${pitftrot},speed=64000000,fps=30"
    fi

    if [ "${pitfttype}" == "28c" ]; then
        overlay="dtoverlay=pitft28-capacitive,rotate=${pitftrot},speed=64000000,fps=30"
    fi

    if [ "${pitfttype}" == "35r" ]; then
        overlay="dtoverlay=pitft35-resistive,rotate=${pitftrot},speed=20000000,fps=20"
    fi
```

where it looks like the ILI9341 based displays get `speed=64000000`, whereas the [HX8357 Rev C-based Pitft 3.5](https://www.adafruit.com/product/2097) gets a `speed=20000000` treatment. This suggests that ILI9341 displays are driven at SPI bus CDIV of 400000000/6400000 rounded up to next even integer, i.e. at `CDIV=8`, which results in effective SPI bus speed of 50mbit/sec, whereas the 3.5 inch PiTFT one gets a `CDIV=400000000/20000000`=20, i.e. an effective SPI bus speed of 20mbits/sec, meaning that the larger 480x320 display will get a 0.4x the speed compared to the smaller 320x240 ones.

Is there a specific reason to configure the 3.5 inch display to use `speed=20000000` here? (was this based on e.g. testing, or known SPI communication rating of HX8357?) I tried to search the spec sheet to find out what the SPI bus speed cap of HX8357 would be, but that has been difficult to find. Do you know what its limit would be?

As a sidenote, I find that I am able to drive your ILI9341-based 2.8 inch resistive 320x240 PiTFT display at a stellar [73.50MHz SPI bus rate](https://github.com/juj/fbcp-ili9341/blob/master/ili9341.h#L16), which is amazing! I have been playing around with another Waveshare ILI9486-based 480x320 resolution display, which [caps out already at a measly 32.25MHz](https://github.com/juj/fbcp-ili9341/blob/master/waveshare35b.h#L12) in comparison! Now I'm pondering whether this is a property of Adafruit vs Waveshare manufacturing, or a property of ILI9341 vs ILI9486 controllers; and whether a HX8357 might be known to have better SPI performance than a ILI9486?

ladyada, 2018-05-15:
yes the HX does not go faster than 20MHz tested. ILI can be overdriven better, you can try tweaking it higher, we like to stick to safe values :)

closed 2018-05-15 by ladyada

ladyada, 2018-05-15:
our fork of fbcp does change-windowing so you dont have to blit the whole display. but yes, SPI is really not designed for what we are doing here and driving it at 80mhz is well beyond expectations :)

juj, 2018-06-17:
Debriefing, I did end up getting an Adafruit HX8357D, couldn't resist the curiosity to see how well it'd work.

The result was that my copy maxes out at 52.33MHz max SPI bus speed, NES Super Mario 3 runs nice at 60fps, and full motion video caps out at 21.29fps, running on my [fbcp-ili9341](https://github.com/juj/fbcp-ili9341) display driver.

I have not yet recorded a video with that display, but here is for reference a video from similar WaveShare 3.5" 320x480 display, https://www.youtube.com/watch?v=dqOLIHOjLq4 , which runs at only 32.25MHz max, so turns out my Adafruit HX8357D display is able to push pixels about +62.26% faster than my WaveShare ILI9486 SPI display. Neat!

If you have a moment to test out fbcp-ili9341 on one or more of your copies of HX8357D, it would be interesting to hear back how much copy to copy variance there is to be expected? Also, tried searching through the Himax spec sheet what is the max stated SPI clock speed of the HX8357D? Oddly it does not seem to have such a statement (or I did not understand to decipher the spec sheet), looks like it only mentions max pixel clocks for the parallel bus modes.

ladyada, 2018-06-17:
congrats! there *is* variation from display to display.
datasheet reference is here
[image]

tWC write cycle = 66 ns -> 15.15MHz