adafruit/Raspberry-Pi-Installer-Scripts #27: Appears fbcp isn't running at boot

joedanks, 2018-06-08:
I'm using a RaspberryPi Zero W with a PiTFT 2.8 capacitive.
Running Raspbian 9.4
I have run the adafruit-pitft script multiple times.  After the first reboot the display turns on and works great.  But any reboot after and I just get a black screen.  I've got it set to mirror the hdmi output.  I can still see the desktop from the hdmi and vnc.  The touch is also working when the screen is black.
I can run `fbcp` after it starts and the display will fire right up.

I added an `echo` line to the `rc.local` file to ensure that file is running at boot and can confirm it does.

Any suggestions to further troubleshoot this?

As a work around, I've got a python script added to the rc.local that listens to gpio to run the fbcp command. Which works for now.

melissagirl, 2018-07-19:
Running `journalctl -xe` gives some more details about the problem. There's a line that says "Unable to open secondary display". My guess as to the nature of the problem is that it's attempting to start fbcp before the display has been properly initialized.

melissagirl, 2018-07-20:
Ok, I found a fix for this issue. Going off my premise that it wasn't being initialized properly first, I added the following to /etc/modules:
`spi_bcm2835`

My suspicion as to why this worked is because the Raspberry Pi Zero W runs with a Broadcom 2835 chip and another SPI module may have been attempted to be loaded. By placing in the /etc/modules, this forced SPI to load first. If you have a different Raspberry Pi, you can run `modprobe spi_bcmXXXX` for the appropriate chip to verify the correct SPI module exists.

brennen, 2018-07-25:
Missed this issue in the shuffle, but I'll take a look this evening.  Thanks for the forensics, @melissagirl.

Arielhh, 2018-07-27:
Adding modprobe "spi_bcm2835" did not help in my case (same pi zero w) any other ideas?

Arielhh, 2018-07-27:
running modprobe spi_bcm2835 from command line - no response. Adding spi_bcm2835 to etc/modules  same issue. I am using pitft 2.8" capacitive from Adafruit.

melissagirl, 2018-07-27:
No response is good. My modules file looks like this:
`i2c-dev
spi_bcm2835`

Possibly i2c-dev needs to be loaded too, so maybe try that.

Arielhh, 2018-07-27:
Thanks, you were right, i installed a fresh version and now the screen is responding however it is very sluggish and the picture is fragmented.  see picture.

[image]

melissagirl, 2018-07-27:
After shutting down, check the LCD ribbon cable connection by pulling the ears, reseating it, pushing them back in. Then try it again.

Arielhh, 2018-07-27:
I tried that but no change. it looks like it is not set to the right resolution although i reinstalled the adafruit configuration script  with the right resolution for the 2.8" screen.

melissagirl, 2018-07-27:
Unfortunately, I don't have that exact model of LCD. I'm actually running a 2.8 resistive plugged into the board designed for the 2.4 due to space constraints and when I tried plugging in my 2.8 capacitive into the same board, the display worked fine too. I'm running the latest Raspbian stretch and with a fresh install and installing the script, I was able to get the LCD working from the command line after booting by typing `fbcp` and running it. If you're not able to do that, then it may be a different issue.

Arielhh, 2018-07-30:
when i am booting to command line the picture on the display is ok. when i startx the desktop is booting on the monitor instead of the LCD. what am i doing wrong, i would like to boot to the LCD without being connected to the HDMI.

melissagirl, 2018-07-30:
To get it working from scratch, I used the latest Raspberry Pi image from 6/29/2018 and after updating it, I followed the Easy Install script instructions. I chose N for the first question and Y for the second. After that, I added spi_bcm2835 right under i2c-dev in /etc/modules and rebooted. It came up fine on both the LCD and HDMI at the same time. I hope that helps.

Arielhh, 2018-07-30:
thanks, something is wrong as i did the same for few times and still  the display stays white when the HDMI cable is not connected.  when the cable is connected and i run startx from the pi (from SSH i am getting an error) the desktop in showing only on the monitor. 

following is the error i am getting when trying startx from SSH:

X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux raspberrypi 4.14.52+ #1123 Wed Jun 27 17:05:32 BST 2018 armv6l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:5F:17:DC vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=df3dce08-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait fbcon=map:10 fbcon=font:VGA8x8
Build Date: 18 October 2017  04:55:30PM
xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support) 
Current version of pixman: 0.34.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Mon Jul 30 21:58:07 2018
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.

melissagirl, 2018-07-30:
Something else I forgot to mention is that I'm doing all of the above steps directly on the Pi plugged into HDMI with a keyboard and mouse. I never have to run `startx` from the command line.

Arielhh, 2018-07-30:
ok, i changed rsapi-config parameters and now i can boot without the HDMI connected but the picture on the lcd looks like it is not in the right resolution. can you please share with me the config.txt file you have? any other place that i can change the tft resolution? (i have 2.8" 320-x240 )

Arielhh, 2018-07-30:
mine is exactly the same except of the capacitive instead of resistive. can you boot without the HDMI cable connected?

melissagirl, 2018-07-30:
Yes. That was my intention with this setup.

Arielhh, 2018-07-30:
maybe my lcd is faulty, i will try to replace it. thanks!!!

Arielhh, 2018-08-01:
It was the LCD, the new one is working just fine. thanks for helping.

melissagirl, 2018-08-02:
That's great to hear. You're welcome.

ladyada, 2018-08-19:
closing cause it seems this is resolved :)

closed 2018-08-19 by ladyada