adafruit/Raspberry-Pi-Installer-Scripts #33: Crontab not working

MeMattStone, 2018-10-21:
Apologies in advance if I've missed something really obvious on this one but on two seperate installs I'm having a problem with crontabs, everything works fine until I run the script to setup as read-only, after this none of my cron jobs are running (everything was setup prior to running the Adafruit script) and typing `crontab -e` for the pi user or `sudo crontab -e` to run as root lets me create a crontab but on reboot these crontabs are lost. Obviously when I try to create these crontabs again I have a jumper on GPIO21 so that the the filesystem is mounted as writeable.

ladyada, 2018-10-21:
hiya which script is this you're referring to?

ladyada, 2018-10-22:
hmm, you could check error logs but it sounds like a linux thing not an issue/mistake with the script, so we're not sure how to fix - we haven't done any testing with the script + chron!

MeMattStone, 2018-10-27:
I got to the bottom of this, lines 264-266 move the location of `/var/spool` to the `/tmp` location and crontabs get written to `/var/spool/cron/crontabs` which will get deleted by the read only script. The workaround for this is to use the system cron at `/etc/crontab` which has a slightly different format as you have to specify the user account before the command but otherwise works fine. I'm guessing the longer term fix would be to symlink `/var/spool/cron` to another folder and move and existing contents before deleting the `/var/spool` folder, not sure if it's possible to symlink inside a symlink? If not the system cron is the answer. Hope this helps anyone else who has had this issue with crontabs.

ladyada, 2018-10-27:
@PaintYourDragon make a note in the guide?

ladyada, 2018-11-12:
https://forums.adafruit.com/viewtopic.php?f=8&t=143442

boelle, 2018-11-12:
I have precise the same issue

as a workaround i go in Write mode and create the crontabs and them copy them to /home/pi/

and then i add to  /etc/rc.local before final exit:

sudo cp /home/pi/pi /var/spool/cron/crontabs/pi
sudo cp /home/pi/root /var/spool/cron/crontabs/root

it's a very rough hack at best

i think @MeMattStone is on to something

boelle, 2018-11-13:
if it can get the job done for both root and pi user then i have protest against it

those that allready have the script what would they do?

PaintYourDragon, 2018-11-14:
TBH it's a bit out of my league, being only an occasional Linux user. Really didn't expect the script would be used on anything more advanced than a RetroPie setup or some Halloween eyes.

Short term, I'm going to document this aspect better so others won't be led into believing it's a robust solution to anything. Longer term, if anyone has a solid workaround for the script to handle this better, I'd gladly merge it.

boelle, 2018-11-14:
well the reason i use it is that every other guide failed, and by failed i mean that there was an error at boot i could not figure and it prevented me from ssh to the pi

this was about the ony thing that made it read-only and allowed me access 

no hard feelings from my side. i'm also an occasional Linux user and i know that has it limits. 

maybe @ladyada & Co knows somebody that knows how we could make it better

ladyada, 2018-11-14:
we don't know who could do this in house - if we did we'd have em fix it! if you know someone please point them to this repo :)