adafruit/Raspberry-Pi-Installer-Scripts #41: i2samp.sh alsamixer volume control not working on Raspbian Lite

AwesomeGinger, 2018-12-09:
It is a known issue that software volume control isn't working when using Raspbian Pixel. But it isn't working on Raspbian Lite Stretch either. I did use the installation script with the latest Raspbian Lite image.

AwesomeGinger, 2018-12-09:
Maybe I found a workaround. If I disable the onboard sound before booting for the first time on a freshly deployed Raspbian Lite, everything seems to be working. But I sillt can't get it to work, if the Pi has been booted at least once.

ladyada, 2018-12-09:
it has to be rebooted at least once but we just tried it and it works for sure!

closed 2018-12-09 by ladyada

AwesomeGinger, 2018-12-09:
I am still having the problem sporadically. And did some more investigation. But first some more details. I am using a Pi Zero W. These are the steps to reproduce the problem: 

1. Copy latest Raspbian lite Image to SD
2. Run i2samp.sh
3. Reboot
4. Re-run i2samp.sh
5. Reboot

Now volume control is working 80% of the time. But after shutting down and restarting the pie there is a high chance volume control with alsamixer isn‘t possible any more. Not just that, but also the audio quality gets terrible. I did test multiple pi zeros.

In one of the re-installatins I didn’t enable /dev/zero playback and now the problem seems to be gone. I assume this could be a timing problem during autostart (The first pop during boot occurs way earlier than I can log in via SSH). But that is just speculation at this point. Disabling /dev/zero playback is just fine in my case. If somebody else runs into this problem and wants to keep /dev/zero playback, maybe setting type=idle in the autostart script might help. But I didn’t test that (yet).

ladyada, 2018-12-09:
try rebooting one last time?

AwesomeGinger, 2018-12-10:
> try rebooting one last time?

Doesn't help. As mentioned disabling ` /dev/zero playback` is fine for me. But I guess I won't stay the only one running into this problem, as I did nothing special.

ladyada, 2018-12-11:
this is super weird, we can't reproduce it, but we added a note to the guide!

AhmedBHameed, 2021-01-17:
I'm heading the same issue.

I have `respberry pi zero w` with [Adafruit Speaker Bonnet for Raspberry Pi](https://learn.adafruit.com/adafruit-speaker-bonnet-for-raspberry-pi/raspberry-pi-usage), I want to control sound volume via command line.

I'm using `amixer set PCM 50%` cli to reduce sound volume as it works perfectly fine with my Ubunto but not with respberry pi.

**amixer scontrols**
```text
Simple mixer control 'PCM',0
```

**amixer version**
```text
amixer version 1.2.2
```

**OS:**
```text
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

Any suggestions ?