adafruit/Raspberry-Pi-Installer-Scripts #42: Rotation selection for 2.8" capacitive PiTFT does not rotate touchscreen itself

OutsourcedGuru, 2018-12-13:
Pi: Raspberry Pi 3B
O/S: Raspbian Stretch Lite
Model:  [Adafruit PiTFT Plus 320x240 2.8" TFT + Capacitive Touchscreen - 2423](https://www.adafruit.com/product/2423)
Reference: [Easy Install](https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/easy-install-2) option
Script: adafruit-pitft.sh
Script responses: 3 (2.8" capacitive), 3 (270 degree rotation), n (TFT as console), y (mirror HDMI)
`/boot/config.txt` result:

```
# --- added by adafruit-pitft-helper Wed Dec 12 15:52:54 PST 2018 ---
dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=pitft28-capacitive,rotate=270,speed=64000000,fps=30
# --- end adafruit-pitft-helper Wed Dec 12 15:52:54 PST 2018 ---
```

This results in the screen appearing in landscape mode and oriented with the tactile pushbuttons to the left of the screen. Unfortunately, running `sudo evtest /dev/input/touchscreen` proves that the touchscreen has not be affected by this setup.

After much trial-and-error, I determined that this would be the correct line which your Easy Setup script should do in this case:

```
dtoverlay=pitft28-capacitive,rotate=270,touch-swapxy,touch-invy,speed=64000000,fps=30
```

By fixing your script, you could be helping out a number of people who—like me—have literally each spent ours trying to research and fix this problem.

ladyada, 2018-12-13:
hiya we rotate it in LXDE - rather than relying on evtest - since the touchscreen doesn't work as  'touchscreen' anymore in console mode. what app is this in?

OutsourcedGuru, 2018-12-13:
LXDE might work for Raspbian PIXEL but not for Lite, right?

I have other projects which include PIXEL and the Desktop (X windows, X11, whatever) but this one is for a 3D printer project with a TFT connected to **Raspbian Lite**. **Kivy** will be the menu interface and I can confirm that my suggestion above works with [Kivy](https://github.com/kivy/kivy)'s [pictures](https://github.com/kivy/kivy/tree/master/examples/demo/pictures) demo app as expected. Likewise, evtest now reports the correct values of X/Y when touched. The Desktop is not installed so there's nowhere to tweak the `/etc/X11` stuff.

Are you sure about that "_doesn't work as 'touchscreen' anymore_" assertion? `ls -l /dev/input/touchscreen` returns something like event3 so it feels like the device tree is happy for this.

----

For what it's worth, you're my go-to supplier of stuff since your tutorials are usually top-notch. (I actually shared Colin's RFID video in Mexico City during a big meeting with one of my clients to explain the technology to them.)

If you want, I could supply you enough to write up a Kivy tutorial how-to for the PiTFT and I've got Python to talk to those four tactile buttons as well.

[Here are some screens](https://help.robo3d.com/hc/en-us/articles/115000715191-Connecting-to-Wi-Fi) that were written in Kivy for a different printer. I'll be working up something even better for this new manufacturer.

ladyada, 2018-12-13:
evtest is ok but if you use other programs they dont recognize it as a touchscreen! 

ok we'll add this, may not be immediate but next time we do a sweep of this library!

OutsourcedGuru, 2018-12-13:
You are the awesome.

ladyada, 2018-12-13:
no YOU are the awesome :D

cmsimike, 2019-12-30:
Thank you so much for this! I've had the TFT for over 2 years now and gave up hope ever getting the calibration correct. Recently the need arose to use the touch screen so I gave it another go today and I was lead here.

jerm, 2020-05-10:
For anyone who is still having trouble with this on the pitft+35: I tried the settings, I couldn't get around this issue for life of me, even with the args above, but then noticed @OutsourcedGuru 's was using options: n (TFT as console), y (mirror HDMI)

Once i switched to this (had been using console:y), it worked like a charm, no extra atfs necessary

makermelissa, 2020-08-03:
Ok, looks like this is currently working in landscape, but not in portrait.

PR #107 (2020-08-04): Fix touch on capacitive 2.8 display for portrait modes
Fixes #42 without messing up anything else.
closed 2020-08-04 by ladyada