adafruit/Raspberry-Pi-Installer-Scripts #44: i2samp aplay service

bobthechemist, 2018-12-28:
I'm wondering if the aplay.service created by i2samp.sh needs to be modified to accommodate the bootup speed of a RPi ZeroW.  As written, the aplay service fails regularly on my Zero.  Adding `After=multi-user.target` to the `[Unit]` section seems to resolve this issue.

Additionally, I appended `&` to the executable that is started, which (I think) makes the service more user-friendly if it is manually started.

ladyada, 2018-12-28:
try an `exec pre` line?
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/adafruit-pitft.sh#L403

bobthechemist, 2018-12-28:
Looks like that is a reasonable option as well.  Ideally, we'd know what service aplay is requiring so that the shortest delay can be used.  I've no idea how to do that, though.

makermelissa, 2025-08-26:
Closing as this was converted and tested.

closed 2025-08-26 by makermelissa