adafruit/Raspberry-Pi-Installer-Scripts #46: adafruit-pitft.sh: Fails to install on buster since tslib was renamed to libts0

teozkr, 2019-02-05:
Repro steps:

1. Install Raspbian Stretch (the current stable) from RPi foundation image
2. Upgrade to Buster (current testing): `sed -i s/stretch/buster/g /etc/apt/sources.list /etc/apt/sources.list.d/*.list && apt update && apt dist-upgrade`
3. Try to install the PiTFT drivers using the install script

Expected result:

PiTFT is active after a reboot.

Actual result:

Script fails to install dependencies, because the package `tslib` was not found.

Workaround:

Installation works fine if you replace [line 192](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/75fd19c0d5a82919877e4efb05ece66d39c3b3e3/adafruit-pitft.sh#L192) with:

```
apt-get install -y bc fbi git python-dev python-pip python-smbus python-spidev evtest tslib libts-bin 1> /dev/null || { warning "Apt failed to install software!" && exit 1; }
```

However, this can't be upstreamed since it would just end up breaking Stretch instead.

teozkr, 2019-02-05:
To confuse things further, tslib still seems to exist in Buster, but [only as a source package](https://packages.debian.org/source/testing/tslib) that libts0 and co. are built from.

ladyada, 2019-02-05:
sounds like we will have to update when Buster mainstreams - but probably not best to do yet?

teozkr, 2019-02-05:
Yeah, I guess it's that, a separate branch, or distro sniffing (which I can understand not wanting to get into).

szarapka, 2019-06-30:
This def needs to be updated now!

ladyada, 2019-06-30:
yep, anyone want to try submitting a PR for a Pi 4?

PR #57 (2019-07-02): Fixes PiTFT Installer for Buster
This change attempts to install libts0 for Buster and if that fails, it tries tslib for older versions of Raspbian.
Fixes #46. Fixes #55.
closed 2019-07-02 by ladyada