adafruit/Raspberry-Pi-Installer-Scripts #47: adafruit-pitft.sh  not working because of 404 error in System upade

herisson9, 2019-02-27:
I try to reinstall my Adafruit screen, but when I lauch adafruit-pitft.sh there's a 404 error in update

This script downloads and installs
PiTFT Support using userspace touch
controls and a DTO for display drawing.
one of several configuration files.
Run time of up to 5 minutes. Reboot required!

Select configuration:
1. PiTFT 2.4", 2.8" or 3.2" resistive (240x320)
2. PiTFT 2.2" no touch (240x320)
3. PiTFT 2.8" capacitive touch (240x320)
4. PiTFT 3.5" resistive touch (320x480)
5. Quit without installing

SELECT 1-5: 3
Select rotation:
1. 90 degrees (landscape)
2. 180 degrees (portait)
3. 270 degrees (landscape)
4. 0 degrees (portait)

SELECT 1-4: 4
[PITFT] Checking init system...
Found systemd
/boot is mounted
[PITFT] System update
Updating apt indexes...
.........
W: The repository 'http://apt.adafruit.com/raspbian wheezy Release' does not have a Release file.
E: Impossible de récupérer http://apt.adafruit.com/raspbian/dists/wheezy/main/binary-armhf/Packages  404  Not Found
E: Impossible de récupérer https://apt.adafruit.com/raspbian/dists/wheezy/main/i18n/Translation-fr_CH  404  Not Found
E: Impossible de récupérer https://apt.adafruit.com/raspbian/dists/wheezy/main/i18n/Translation-en  404  Not Found
E: Impossible de récupérer https://apt.adafruit.com/raspbian/dists/wheezy/main/i18n/Translation-fr_CH.UTF-8  404  Not Found
E: Impossible de récupérer https://apt.adafruit.com/raspbian/dists/wheezy/main/i18n/Translation-fr  404  Not Found
E: Le téléchargement de quelques fichiers d'index a échoué, ils ont été ignorés, ou les anciens ont été utilisés à la place.
WARNING : Apt failed to update indexes!

I try to change the line 145 and 192 as I've seen in the other issues, but still not working.

ladyada, 2019-02-27:
are you running raspbian wheezy?

herisson9, 2019-02-27:
No, it's stretch.
So, wich command should I write?

ladyada, 2019-02-27:
i dont know - it has worked for everyone else - you could remove that line from the script?

ladyada, 2019-06-30:
closing from no activity

closed 2019-06-30 by ladyada