adafruit/Raspberry-Pi-Installer-Scripts #50: adafruit-pitft.sh fails with "pip install evdev"

radio24, 2019-05-26:
When I was using `sudo ./adafruit-pitft.sh` all work as expected, until adafruit-pitft.sh fails with `pip install evdev`. I did already `sudo pip install --upgrade pip` and `sudo pip install -U setuptools` but the error is still the same:

> ERROR: Complete output from command /usr/bin/python3 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-7z9w2qqc/evdev/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-u2723sbm/install-record.txt --single-version-externally-managed --compile:
    ERROR: running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.5
    creating build/lib.linux-armv7l-3.5/evdev
    copying evdev/evtest.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/ff.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/device.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/util.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/genecodes.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/uinput.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/ecodes.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/eventio.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/__init__.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/eventio_async.py -> build/lib.linux-armv7l-3.5/evdev
    copying evdev/events.py -> build/lib.linux-armv7l-3.5/evdev
    running build_ext
    running build_ecodes
    writing ecodes.c (using /usr/include/linux/input.h /usr/include/linux/input-event-codes.h /usr/include/linux/uinput.h)
    building 'evdev._input' extension
    creating build/temp.linux-armv7l-3.5
    creating build/temp.linux-armv7l-3.5/evdev
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c evdev/input.c -o build/temp.linux-armv7l-3.5/evdev/input.o -std=c99 -Wno-error=declaration-after-statement
    evdev/input.c:10:20: fatal error: Python.h: No such file or directory
     #include <Python.h>
                        ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command "/usr/bin/python3 -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-7z9w2qqc/evdev/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-u2723sbm/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-7z9w2qqc/evdev/

ladyada, 2019-05-26:
what's the exact SBC and operating system version you're running?

radio24, 2019-05-26:
SBC: Raspberry Pi 3 Model B+
OS: Raspbian light "Stretch" with Kernel 4.19.42 (updated with `sudo apt-get`)

ladyada, 2019-05-26:
Try `sudo apt-get install python3-dev` or similar

radio24, 2019-05-26:
Great! That worked.
Thanks a lot 👍

closed 2019-05-26 by radio24