adafruit/Raspberry-Pi-Installer-Scripts #59: 2.8 Capacitive Touch Rotation not working on Buster

makermelissa, 2019-07-16:
When testing rotation, here's the results:
90 Degrees: Touch is 90 degrees different than rotation
180 Degrees: Touch works normally, but screen only takes up upper 3/4s.
270 Degrees: Touch is 270 degrees different than rotation.
0 Degrees: Touch is 180 degrees different than rotation and screen only takes up upper 3/4s.

So it appears that touch rotation is not working, at least with capacitive touch. I will have to try with other displays, though I believe the 3.5" resistive touch worked fine.

makermelissa, 2019-07-17:
It looks like there is touch calibration data for the 2.8" Capacitive, 3.5", and 2.8" Resistive touch screens. I have the 3.5 and 2.8 Capacitive screens. The 3.5 appears normal, so I've ordered a 2.8" Resistive touch screen and will see how the touch rotation compares on that. If it's only the capacitive one that is having issues, I'll focus on that.

makermelissa, 2019-07-19:
Resistive Touch isn't working at all on the 2.8" and 2.4". I noticed between the Capacitive and Resistive touch screens, there's an extra event under /dev/input so it may be the driver not installing correctly. I'll try the 3.5" again. I thought that one was working and since they both should be using the STMPE610 chip, they should both work.

makermelissa, 2019-07-19:
More oddities. 
1. 2.8" resistive screen does not work on Buster, but does on Stretch. 
2. 3.5" touchscreen is working with the 2.8" resistive touch driver
3. 3.5" touchscreen also work's with its own driver.

varosi, 2019-08-08:
I have the same experience on 2.8 Capacitive display on latest Buster.

JenusL, 2019-12-03:
I have the 3.5" resistive touch display from Adafruit and the rotation is worng for me. I have it set to 270 degrees but the touch is + or - 90 degrees wrong. Does anyone have any idea of what could be wrong?

JenusL, 2019-12-03:
Ok I found the issue for me. The X11 calibration was never installed for me and here's why.
The script is looking for `/etc/lightdm` to determine if X11 should be configured or not. The problem is that I don't have LightDM installed and don't need it. But I have X11 installed and I need the calibration.
So we need a better way to check if X11 is installed. Maybe just check if `/etc/X11` exist instead of lightdm?

makermelissa, 2019-12-03:
Thanks @JenusL. I'll have to take another look at the script. That'd be awesome if it's that easy.

JenusL, 2019-12-04:
Remember that the script uses `/etc/lightdm` to check for X11 at three different places in the code.
For calibration, auto-login to desktop and resolution scaling. I only needed the calibration so I skipped the other two.

svenyonson, 2019-12-30:
Thank God for Google. Some of these problems are so arcane that you would never find the solutions on your own. I found the fix for the rotation issue (at least in my case)[here:](https://www.0xf8.org/2016/01/complete-rotation-support-for-the-adafruit-pitft-2-8-capacitive-touchscreen-display/)
TLDR: You need the second line in your /boot/config.txt file:
`dtoverlay=pitft28-capacitive,rotate=90,speed=64000000,fps=30
dtoverlay=pitft28-capacitive,touch-swapxy=true,touch-invx=true
`
Note it has to be on two lines because of a line length limit in the script processing. 
Hope this helps your issue too.

makermelissa, 2020-03-09:
@svenyonson, I love your solution. Playing around with the parameters, I was able to get it adjusted on all rotations. I will add setting the correct parameters in the script.

PR #76 (2020-03-09): Fixes Capacitive screen rotation in buster
Fixes #59. Note that screen scaling appears to be off (like either the touch input area is smaller than the display or perhaps the display just isn't showing the full desktop), but at least the mouse pointer moves in the correct direction now, which is a definite improvement.
closed 2020-03-09 by makermelissa