adafruit/Raspberry-Pi-Installer-Scripts #64: I2S aplay /dev/zero script blocks playback: "Device or resource busy"

besi, 2019-11-03:
I did run the installer script for the I2S Class D amp and everything did work fine right from the start. 

However, if I do enable the option *activate '/dev/zero' playback* [1](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/i2samp.sh#L450)
then I get an error when attempting to playback some audio:

> aplay: main:788: audio open error: Device or resource busy

Could it be that only on instance can be using aplay at any given moment?

ladyada, 2019-11-03:
no, you can def run more than once. try rebooting

edmcdonagh, 2020-10-26:
I have this same issue - unless I kill the systemd aplay service I can't play anything - `aplay: main:828: audio open error: Device or resource busy`

I have rebooted multiple times, but the only solution so far is to kill the service or to disable it.

The same is true if I try an aplay two wav files simultaneously - what is it that makes aplay accept two inputs?

I'd be grateful for any pointers!

makermelissa, 2025-08-26:
Closing. This was recently updated and tested.