adafruit/Raspberry-Pi-Installer-Scripts #69: adafruit-pitft: Rotation not working

FD-, 2020-02-05:
I cannot seem to get my new 240x240 PiTFT Mini to display with 270deg rotation, so that the text is readable from the side the header is mounted (with the two buttons to the right of the display).

I initially set up the display with 90 deg rotation (normal landscape), and, upon realising I needed reverse landscape, I re-ran the script, choosing 270 deg this time. Still, I only saw normal landscape, so I assumed the script cannot be run again for adjusting the settings, and instead started from a fresh Raspbian Lite image. However, I still only get landscape when choosing 270 deg rotation.

ladyada, 2020-02-05:
try editing the rotation directly in /boot/config.txt!

FD-, 2020-02-05:
I tried lcd_rotate=2, display_hdmi_rotate=2 and display_lcd_rotate=2, and none worked, unsurprising to me, since I assume they modify GPU rendering, but I don’t think the console-only Kernel driver goes through the GPU.
Just to clarify: This is about the console driver, not about HDMI mirroring (which doesn‘t work at all as described in #71 )
Your comment also sounded as if the config already contained a rotation value that I just had to edit, which wasn‘t the case. The Script seems to just add entries that enable SPI.

FD-, 2020-02-05:
I also tried adding rotate=180 to /etc/modprobe.d/fbtft.conf (to the fbtft_device options) to no avail, so please reopen this issue.

makermelissa, 2020-02-05:
Based on your other issues, I believe you are trying to display console only, which currently doesn't support rotation. You might find some more useful info in #42, which appears to have some info on accomplishing this.

makermelissa, 2020-02-05:
I was corrected, apparently the console should rotate. It looks like the place to change it for this display is probably `/etc/modprobe.d/fbtft.conf`.

makermelissa, 2020-02-05:
Can you try adding rotate=2 to `/etc/modprobe.d/fbtft.conf`?

FD-, 2020-02-05:
I agree from reading through the script that it doesn't actually support setting a rotation for the console-only driver, but from a user perspective: Why am I asked about rotation when it actually doesn't work? I obviously didn't read the script before using it, but I would assume the majority of users doesn't.

Anyway, I've found a [very useful blog post](https://krystof.io/mirroring-raspberry-pi-hdmi-video-to-a-st7789-1-3-inch-lcd-display/) about using the ST7789V driver chip on a Raspberry Pi, and I'm very close to having rotation working:

By replacing the content of /etc/modprobe.d/fbtft.conf with

```
options fbtft_device name=fb_st7789v custom width=240 height=240 gpios=dc:25,cs:8,led:26 speed=40000000 bgr=1 fps=60
options flexfb setaddrwin=0 width=240 height=240 init=-1,0x11,-2,120,-1,0x36,0x00,-1,0x37,0x00,0x50,-1,0x3A,0x05,-1,0xB2,0x0C,0x0C,0x00,0x33,0x33,-1,0xB7,0x35,-1,0xBB,0x1A,-1,0xC0,0x$
```
 rotation works fine, but it seems the colors change: The end of the command prompt (~$) is now red instead of blue!

What the above does is AFAICT it chooses to use the fb_st7789v driver instead of the flexfb driver for the display. Apparently, the st7789v driver supports rotation, while the flexfb driver doesn't (it just ignores the argument). The 'custom' argument and size specification in the fbtft_device options and the additional 0x37 command in the flexfb init option are required, otherwise we only get a black screen, the bottom 240x240 pixels of a 240x320 area or the top of that area respectively. I still don't know how flexfb plays into the whole thing, since the blog post I linked above doesn't use it.

@makermelissa Are you sure you're not mixing up /boot/config.txt and

makermelissa, 2020-02-05:
You're correct. I'm playing around with the init codes and will give you something when I can get it working as you are wanting.

FD-, 2020-02-05:
I think I got it: It was a simple matter of setting bgr=0.
Full config (btw, I think my previous listing was flawed, I lost some parts when copy-pasting).

```
options fbtft_device name=fb_st7789v custom width=240 height=240 gpios=dc:25,cs:8,led:26 speed=40000000 bgr=0 fps=60 rotate=180 init=-1,0x11,-2,120,-1,0x36,0x00,-1,0x37,0x00,0x50,-1,0x3A,0x05,-1,0xB2,0x0C,0x0C,0x00,0x33,0x33,-1,0xB7,0x35,-1,0xBB,0x1A,-1,0xC0,0x2C,-1,0xC2,0x01,-1,0xC3,0x0B,-1,0xC4,0x20,-1,0xC6,0x0F,-1,0xD0,0xA4,0xA1,-1,0x21,-1,0xE0,0x00,0x19,0x1E,0x0A,0x09,0x15,0x3D,0x44,0x51,0x12,0x03,0x00,0x3F,0x3F,-1,0xE1,0x00,0x18,0x1E,0x0A,0x09,0x25,0x3F,0x43,0x52,0x33,0x03,0x00,0x3F,0x3F,-1,0x29,-3
options flexfb setaddrwin=0 width=240 height=240
```

makermelissa, 2020-02-05:
Excellent, I'll test it out.

makermelissa, 2020-02-05:
Thanks, seems to be working. I'll see if I can get the script working to make use of the rotate a bit later.

makermelissa, 2020-02-05:
This code also works and is closer to the original:
```
options fbtft_device name=flexfb gpios=dc:25,cs:8,led:26 speed=40000000 bgr=1 fps=60 rotate=180
options flexfb setaddrwin=0 width=240 height=240 init=-1,0x11,-2,120,-1,0x36,0xC0,-1,0x37,0x00,0x50,-1,0x3A,0x05,-1,0xB2,0x0C,0x0C,0x00,0x33,0x33,-1,0xB7,0x35,-1,0xBB,0x1A,-1,0xC0,0x2C,-1,0xC2,0x01,-1,0xC3,0x0B,-1,0xC4,0x20,-1,0xC6,0x0F,-1,0xD0,0xA4,0xA1,-1,0x21,-1,0xE0,0x00,0x19,0x1E,0x0A,0x09,0x15,0x3D,0x44,0x51,0x12,0x03,0x00,0x3F,0x3F,-1,0xE1,0x00,0x18,0x1E,0x0A,0x09,0x25,0x3F,0x43,0x52,0x33,0x03,0x00,0x3F,0x3F,-1,0x29,-3
```
I was running into the issue of it being offset, but adding `0x37,0x00,0x50,-1,` fixed that.

FD-, 2020-02-05:
Does that mean you saw an effect after just adding rotate=180 or did you also make changes to the into code beyond adding the 0x37 command?

makermelissa, 2020-02-05:
Oh right, no the `rotate=180` was not working for me. The rotation is done via changes to the init code only. I edited the above to match my Pi.

makermelissa, 2020-02-05:
My plan is to choose the rotation init code from 1 of 4 values depending on the rotation setting and inject that into the init sequence.

PR #73 (2020-02-06): Added 240x240 MADCTL injection based on rotation selection
This allow console rotation on the 240x240 display. Fixes #69.
closed 2020-02-06 by makermelissa