adafruit/Raspberry-Pi-Installer-Scripts #70: adafruit-pitft: fbtft_device: spi_busnum_to_master(0) returned NULL

FD-, 2020-02-05:
When running the setup script for my new 240x240 PiTFT Mini, the display stays black when rebooting after the script has run through. From dmesg, I see that the fbtft_device driver prints error messages:

[    8.038705] fbtft_device: spi_busnum_to_master(0) returned NULL
[    8.041871] fbtft_device: failed to register SPI device
[    8.167084] systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
[    8.215491] systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
[    8.268404] systemd[1]: Failed to start Load Kernel Modules.

After some googling, I found the problem is the order in which the modules are added to /etc/modules. If I manually change spi-bcm2835 to be the first in the list, the display starts to work.

ladyada, 2020-02-05:
thats odd, we just tried it on a fresh pi a week ago, did you have a new raspbian install?

FD-, 2020-02-05:
I was using the most recent Raspbian image from the Raspberry Pi website (I think it’s from September). Also, I forgot to mention I’m using a Pi Zero W.

makermelissa, 2020-02-05:
Hello, I'm gonna take a look at this.

makermelissa, 2020-02-05:
Ok, I was able to reproduce. This sounds suspiciously like #27, which I found only affected the Raspberry Pi Zero W (possibly the Zero too). That issue might have been what you found while googling as I was the one who figured out about adding `spi-bcm2835` to `/etc/modules` as the fix.

FD-, 2020-02-05:
Actually, the fix was trivial to find just by Googling the error message "spi_busnum_to_master(0) returned NULL", which shows up in dmesg. The error isn't specific to the Adafruit product but to the fbtft project the script installs behind the scenes. Has been discussed there as early as 2015: https://github.com/notro/fbtft/issues/227.

closed 2020-02-05 by ladyada

makermelissa, 2020-02-05:
I updated the script to put spi-bcm2835 in the list first and tested that it works by re-running the script.

PR #72 (2020-02-05): Moved spi-bcm2835 to be first in /etc/modules
Fixes #70