adafruit/Raspberry-Pi-Installer-Scripts #75: adafruit-pitft.sh: "Unable to locate package python-spidev" in Buster Lite

9V1LH, 2020-02-26:
I'm using Raspbian Buster Lite, release date 2020-02-13

[PITFT] Installing Python libraries & Software...
Installing Pre-requisite Software...This may take a few minutes!
E: Unable to locate package python-spidev
WARNING : Apt failed to install software!

ladyada, 2020-02-26:
make sure you have internet connectivity and are running python3

9V1LH, 2020-02-27:
I haven't installed python3, but this quick and dirty workaround did the trick for me:

$sudo apt install -y python-pip
$pip install spidev

Then I removed "python-pip" from line 199 and run the script

ladyada, 2020-02-27:
ok thanks! not sure why only you had this issue (we have lots of folks use this code :)

closed 2020-02-27 by ladyada

aanrudolph2, 2021-09-01:
Seeing the same issue as OP. I'm running Ubuntu 20.10; apt-get couldn't find python-spidev when I ran it manually.

Additionally, the overlays folder in Ubuntu are in a different location: /boot/firmware/overlays. For now I can get it working by altering the script. I am intentionally not opening an issue because I realize this may not be supported.

tbjers, 2021-12-09:
I ran into this issue as well on Rasbian Buster (though on a Kiwix image, which I think is based on Pi OS Lite), and for me the way to fix this was to completely replace the contents of `/etc/apt/sources.list` with the following:

```
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
deb http://archive.raspberrypi.org/debian/ buster main

# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
```

And wipe my cache by running:
```bash
$ rm -Rf /var/cache/apt
```