adafruit/Raspberry-Pi-Installer-Scripts #78: touchscreen calibration

uglyoldbob, 2020-03-24:
Hello,

I am using the 2.8 " touchscreen with a 4gb raspberry pi 4.
https://www.adafruit.com/product/1601

I ran the adafruit-pitft.sh, option 1, then option 3, then 'n', then 'y'.

The touchscreen appears to think the display is 640x240 instead of 320x240. Up and down seems to be accurate, but left and right appears to be scale by 2.

How do I fix this?

uglyoldbob, 2020-03-26:
I managed to at least temporarily solve my problem by issuing the command "DISPLAY=:0 xrandr -s 640x480". For some reason the display was at 1280x480.

    pi@raspberrypi:~ $ DISPLAY=:0 xrandr
    Screen 0: minimum 320 x 200, current 1280 x 480, maximum 7680 x 7680
    HDMI-1 connected primary 1280x480+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
       FIXED_MODE    59.38*+
       640x480       58.91

After i ran the command:

    pi@raspberrypi:~ $ DISPLAY=:0 xrandr                                            
    Screen 0: minimum 320 x 200, current 640 x 480, maximum 7680 x 7680
    HDMI-1 connected primary 640x480+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
       FIXED_MODE    59.38 +
       640x480       58.91*

makermelissa, 2020-08-04:
I just tested and this appears to be resolved with the recent refactoring.

closed 2020-08-04 by makermelissa