adafruit/Raspberry-Pi-Installer-Scripts #85: PiTFT: Uninstall Option

makermelissa, 2020-04-22:
In order to make it easier for users to remove the kernel level driver, it would be great to have an uninstall option that went through and cleaned up everything that it installed so users could use it with Python.

Trudko, 2020-05-09:
@makermelissa Do you please have an idea, what would be workaround till then? I've installed kernel driver but would love to use it with python now.

makermelissa, 2020-05-10:
Hi. Look in /boot/config.txt. You’ll need to remove the appropriate pitft device tree overlay for now.

makermelissa, 2020-05-12:
Ok, I had to do this manually on one of my Raspberry Pis today and I had to remove some lines from `/etc/modules` as well.

Trudko, 2020-05-13:
@makermelissa thank you, could you maybe go more into details what exactly have you changed?

makermelissa, 2020-05-16:
Fixed via #90

makermelissa, 2020-05-16:
It's available, so you can look at the script now @Trudko  ;)

closed 2020-05-16 by makermelissa