adafruit/Raspberry-Pi-Installer-Scripts #87: PiTFT 3.5" Resistive touch rotation not working

gazzmanzx6, 2020-04-27:
With Raspbian Buster the touch rotation in not working when I rotate the screen by 270 degrees.
All touch movement is inverted.
I've tried adding touch-swapxy,touch-invy to the  /boot/config.txt file but it hasn't helped.
Is there anyway to fix this?

makermelissa, 2020-08-03:
Fixed via #104. Tested in all 4 rotations.

closed 2020-08-03 by makermelissa

gazzmanzx6, 2020-08-04:
I now get a completely blank screen after updating. The Pi has started up ok as I can access the desktop via VNC.

ladyada, 2020-08-04:
after updating, you may need to run the installer script again

gazzmanzx6, 2020-08-04:
I have run the installer multiple times but always get a blank screen. I’ve even tried uninstalling and reinstalling but no luck.

ladyada, 2020-08-04:
you can post the output, or try a fresh copy of raspbian - check your wiring and seating as well. we just tested all screens with latest raspi OS

gazzmanzx6, 2020-08-04:
Ok. Fixed it. Had to update Raspbian and then reinstall PiTFT. All working correctly now. Thanks.

makermelissa, 2020-08-04:
Excellent

gazzmanzx6, 2020-08-04:
Thanks for all your work.