adafruit/Raspberry-Pi-Installer-Scripts #89: pitft-fbcp.sh failed on Ubuntu

windoze, 2020-04-27:
On RPi4B/Ubuntu-arm64, the script failed with the error
```
/home/ubuntu/repos/rpi-fbcp/main.c:9:10: fatal error: bcm_host.h: No such file or directory
```
This file is part of Raspbian, but seems it doesn't exist in Ubuntu.

Is there anyway I can use the screen in Ubuntu-arm64?'

Thanks.

ladyada, 2020-04-27:
not really - the software is tuned for raspbian kernel, if you find a solution you can submit a PR

craiglink, 2021-03-10:
```sudo apt-get install libraspberrypi-dev```

will fix the issue

ezraholm50, 2021-03-15:
Also had to install build-essential on DietPi. PR submitted.
Thanks.

makermelissa, 2025-08-26:
Closing as a bigger issue has eclipsed this and doesn't currently work with bookworm. See #295.

closed 2025-08-26 by makermelissa