adafruit/Raspberry-Pi-Installer-Scripts #91: Old vesrion of adafruit-pitft.sh ?

kokein, 2020-05-17:
Hello, 

There is something wrong with this new installer. 
I have TFT 2.2" 320×240 (No Touch) thet is no longer able to install with "easy" script. 
I'm getting black/gray lines screen. 
Can You share an old version? It was working about three weeks ago :(

kokein, 2020-05-17:
Ok i found it working when go back to this version (below) - but "uninsall" did not worked. 
I had to use clean install of distro (raspbian) 
Tested: https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/94f147b64f36721de59f48cbb479eadf876cfafe/adafruit-pitft.sh

makermelissa, 2020-05-18:
Please provide some more specifics about what options you chose. I just tested with a 2.2" PiTFT with a fresh install on Raspian Full using Screen mirroring and did not experience the symptoms you described. I don't doubt that you are seeing the issues you describe, but I would like to reproduce them so I can figure out where the issue is happening.

Which specific Raspbian image did you experience the issue with?
Is the issue happening with the install, uninstall, or both?
Did you display the console?
Did you choose the screen mirroring option?
Did you reboot after?

Thanks for letting us know.

kokein, 2020-05-18:
Sure :)
It was Raspbian Lite - clan intsall. (hash checked)
I did use new script as mentioned, but after script finished and reboot
those "stipes" appeared.
It was a actually black screen and some brighter (gray) bars horizontal.
So i did an "uninstall" script. Did not work (noting happen)
Went back to older installer - nothing helped, and did one more time c
clean install (from the same image)
But lunched an older one version (since month ago) and my LCD "woke up" and
now it's working ok.

Which specific Raspbian image did you experience the issue with?
https://www.raspberrypi.org/downloads/raspbian/  *February 2020*
Is the issue happening with the install, uninstall, or both? *both*.
Did you display the console? i'm working on ssh but using lcd as a monotor
console (it was only black/gray bars)
Did you choose the screen mirroring option? ?! 270 degrees rotation option
Did you reboot after?  sure - i've done it several times till tomorrow ,
that was no problems on older scripts.

Nick :)

makermelissa, 2020-05-18:
Thanks. I'll give it a try with those options.

makermelissa, 2020-05-18:
I am still unable to reproduce this. After choosing the install options:
[image]
After uninstalling:
[image]

kokein, 2020-05-18:
I dont know :(

makermelissa, 2020-05-18:
Me neither. I'm going to leave this open for now in case anybody else experiences this and we can get more information. Thanks for your help.

makermelissa, 2020-07-27:
Closing this because I haven't heard anything from anybody else and have been unable to reproduce.

closed 2020-07-27 by makermelissa