adafruit/Raspberry-Pi-Installer-Scripts #92: NTP time does not sync (Temporary failure in name resolution)

SkyperTHC, 2020-06-05:
ntpd is running correctly but does not sync time. Does not happen on every reboot but every once in a while.
root@PoolPi:/home/pi# ps alxww | grep ntpd
5   109   411     1  20   0   7960  3260 poll_s Ssl  ?          0:00 /usr/sbin/ntpd -p /var/run/ntpd.pid -g -u 109:114
---------------------
root@PoolPi:/home/pi# strace -s180 -p 411
...
clock_gettime(CLOCK_REALTIME, {tv_sec=1550139211, tv_nsec=743923000}) = 0
read(23, "\0", 32)                      = 1
gettimeofday({tv_sec=1550139211, tv_usec=745815}, NULL) = 0
getpid()                                = 411
send(3, "<27>**Feb 14 10:13:31 ntpd[411]: error resolving pool 1.debian.pool.ntp.org: Temporary failure in name resolution (-3)", 116, MSG_NOSIGNAL) = 116**
...
----------------------
It goes into an endless loop through all NTP pool servers and reports the same error for every one of the configured pool servers.

However, DNS is working fine from the shell and 1.debianpool.ntp.org resolves nicely:
$ ping 1.debian.pool.ntp.org
PING 1.debian.pool.ntp.org (92.68.81.154) 56(84) bytes of data.
64 bytes from www.elandsgracht.nl (92.68.81.154): icmp_seq=1 ttl=50 time=26.2 ms

I suspect that ntpd binds to the wrong interface (??) and never tries to re-bind. It really should be fixed in ntpd but a quick workaround is to force ntpd in /etc/rc.local just before 'exit' call:
/etc/init.d/ntp stop
sleep 5
ntpd -qng >/dev/null 2>/dev/null
/etc/init.d/ntp start

SkyperTHC, 2020-06-06:
It only happens if ntp is started via 'systemctl' (which is the default). It works fine when started directly from the command line.

It appears that the problem is the sym link from /etc/resolv.conf to /tmp/dhcpcd.resolv.conf

When started from 'systemctl start ntp' then ntpd fails to open that file:
`stat64("/etc/resolv.conf", 0xbea7af80) = -1 ENOENT (No such file or directory)`

However, the file truly exists and is readable by uid=109 (ntp user):
```
root@Zero2:/ # file /etc/resolv.conf
/etc/resolv.conf: symbolic link to /tmp/dhcpcd.resolv.conf
root@Zero2:/ # ls -al /etc/resolv.conf
lrwxrwxrwx 1 root root 23 Jun  6 14:45 /etc/resolv.conf -> /tmp/dhcpcd.resolv.conf
root@Zero2:/etc # ls -ald /tmp/
drwxrwxrwt 8 root root 180 Jun  6 14:39 /tmp/
root@Zero2:/etc # ls -ald /tmp/dhcpcd.resolv.conf
-rw-r--r-- 1 root root 49 Feb 14  2019 /tmp/dhcpcd.resolv.conf
```

Perhaps systemctl does some CAPs kong-fu or does some chroot funk. No patience to debug this. Workaround is to remove the symlink and hard-code the nameserver:

```
mount -o remount,rw /
rm /etc/resolv.conf
cp /tmp/dhcpcd.resolv.conf /etc/resolv.conf
mount -o remount,ro /
```

nabelekt, 2022-04-30:
I am having the same or a similiar issue.

```
ubuntu@ubuntu:~$ date
Mon Jan 10 04:59:58 UTC 2022
ubuntu@ubuntu:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.3 LTS
Release:        20.04
Codename:       focal

ubuntu@ubuntu:~$ ntpd --version
ntpd 4.2.8p12@1.3728-o (1)
```

I have this script which is triggered by a systemctl service once a network econnection is established on boot:
```
#!/bin/bash

# Trigerred by ntp-sync.service at boot once a network connection is established/the network interface comes up
# Force an NTP sync

LOG_FILE_PATH='/home/ubuntu/logs/ntp_sync.log'

# Check Bash version
if [ -z "${BASH_VERSINFO}" ] || [ -z "${BASH_VERSINFO[0]}" ] || [ ${BASH_VERSINFO[0]} -lt 5 ]; then
    echo "This script requires Bash version >= 5. Exiting."
    exit 1
fi

echo "STARTING: $(date +'%F %T'): uptime: $(awk '{print int($1/3600)":"int(($1%3600)/60)":"int($1%60)}' /proc/uptime)" >> $LOG_FILE_PATH

# Taken from https://askubuntu.com/a/256004/570541
service ntp stop >> $LOG_FILE_PATH 2>&1
ntpd -gq >> $LOG_FILE_PATH 2>&1
service ntp start >> $LOG_FILE_PATH 2>&1

echo "DONE: $(date +'%F %T'): uptime: $(awk '{print int($1/3600)":"int(($1%3600)/60)":"int($1%60)}' /proc/uptime)" >> $LOG_FILE_PATH
```

Here is the content of the log generated by that script starting with the last couple lines before the last reboot which was now about 2 days 15 hr ago.

<details>
  <summary>Log contents</summary>
  
  ```
  10 Jan 06:02:41 ntpd[942]: Soliciting pool server 94.224.67.24
  23 Apr 01:56:16 ntpd[942]: ntpd: time set +8884413.018196 s
  ntpd: time set +8884413.018196s
  DONE: 2022-04-23 01:56:24: uptime: 1:6:19
  STARTING: 2022-01-10 04:56:47: uptime: 0:0:15
  10 Jan 04:56:56 ntpd[940]: ntpd 4.2.8p12@1.3728-o (1): Starting
  10 Jan 04:56:56 ntpd[940]: Command line: ntpd -gq
  10 Jan 04:56:56 ntpd[940]: proto: precision = 0.148 usec (-23)
  10 Jan 04:56:56 ntpd[940]: leapsecond file ('/usr/share/zoneinfo/leap-seconds.list'): good hash signature
  10 Jan 04:56:56 ntpd[940]: leapsecond file ('/usr/share/zoneinfo/leap-seconds.list'): loaded, expire=2022-12-28T00:00:00Z last=2017-01-01T00:00:00Z ofs=37
  10 Jan 04:56:56 ntpd[940]: Listen and drop on 0 v6wildcard [::]:123
  10 Jan 04:56:56 ntpd[940]: Listen and drop on 1 v4wildcard 0.0.0.0:123
  10 Jan 04:56:56 ntpd[940]: Listen normally on 2 lo 127.0.0.1:123
  10 Jan 04:56:56 ntpd[940]: Listen normally on 3 lo [::1]:123
  10 Jan 04:56:56 ntpd[940]: Listening on routing socket on fd #20 for interface updates
  10 Jan 04:56:57 ntpd[940]: error resolving pool 0.ubuntu.pool.ntp.org: Temporary failure in name resolution (-3)
  10 Jan 04:56:58 ntpd[940]: error resolving pool 1.ubuntu.pool.ntp.org: Temporary failure in name resolution (-3)
  10 Jan 04:56:59 ntpd[940]: error resolving pool 2.ubuntu.pool.ntp.org: Temporary failure in name resolution (-3)
  10 Jan 04:57:00 ntpd[940]: error resolving pool 3.ubuntu.pool.ntp.org: Temporary failure in name resolution (-3)
  10 Jan 04:57:01 ntpd[940]: error resolving pool ntp.ubuntu.com: Temporary failure in name resolution (-3)
  10 Jan 04:57:09 ntpd[940]: Listen normally on 4 eth0 [fe80::e65f:1ff:fe68:6a1e%2]:123
  10 Jan 04:57:09 ntpd[940]: new interface(s) found: waking up resolver
  10 Jan 04:57:18 ntpd[940]: Deleting interface #4 eth0, fe80::e65f:1ff:fe68:6a1e%2#123, interface stats: received=0, sent=0, dropped=0, active_time=9 secs
[... 460 lines cut ...]
  10 Jan 06:36:17 ntpd[940]: Soliciting pool server 209.50.63.74
  10 Jan 06:36:23 ntpd[940]: Soliciting pool server 2001:67c:1560:8003::c8
  10 Jan 06:36:35 ntpd[940]: Soliciting pool server 129.250.35.250
  10 Jan 06:37:09 ntpd[940]: Soliciting pool server 173.0.48.220
  10 Jan 06:37:15 ntpd[940]: Soliciting pool server 45.79.51.42
  10 Jan 06:37:24 ntpd[940]: Soliciting pool server 204.2.134.163
  ...
  12 Jan 20:19:07 ntpd[940]: Soliciting pool server 172.107.84.94
  12 Jan 20:19:15 ntpd[940]: Soliciting pool server 108.61.56.35
  12 Jan 20:19:18 ntpd[940]: Soliciting pool server 207.192.69.118
  12 Jan 20:19:44 ntpd[940]: Soliciting pool server 91.189.94.4
  12 Jan 20:19:54 ntpd[940]: Soliciting pool server 2001:41d0:8:4d0d::1
  12 Jan 20:20:14 ntpd[940]: Soliciting pool server 162.209.90.58
  12 Jan 20:20:19 ntpd[940]: Soliciting pool server 147.182.158.78
  12 Jan 20:20:24 ntpd[940]: Soliciting pool server 216.240.36.24
  12 Jan 20:20:49 ntpd[940]: Soliciting pool server 91.189.89.198
  12 Jan 20:21:00 ntpd[940]: Soliciting pool server 45.63.54.13
  12 Jan 20:21:18 ntpd[940]: Soliciting pool server 144.76.159.151
  12 Jan 20:21:23 ntpd[940]: Soliciting pool server 50.205.244.109
  12 Jan 20:21:29 ntpd[940]: Soliciting pool server 108.61.219.13
  12 Jan 20:21:55 ntpd[940]: Soliciting pool server 91.189.89.199
  12 Jan 20:22:04 ntpd[940]: Soliciting pool server 45.33.68.112
  ```
</details>
Time has not synced since reboot, but has an Internet connection has been able to previously sync after a while.

nabelekt, 2022-04-30:
Following the suggestion above to set `PrivateTmp=false` seems to have resolved the `error resolving pool 0.ubuntu.pool.ntp.org: Temporary failure in name resolution (-3)` errors, but it still doesn't seem to want to sync. I now see
```
STARTING: 2022-01-10 04:56:47: uptime: 0:0:15
10 Jan 04:56:56 ntpd[939]: ntpd 4.2.8p12@1.3728-o (1): Starting
10 Jan 04:56:56 ntpd[939]: Command line: ntpd -gq
10 Jan 04:56:56 ntpd[939]: proto: precision = 0.148 usec (-23)
10 Jan 04:56:56 ntpd[939]: leapsecond file ('/usr/share/zoneinfo/leap-seconds.list'): good hash signature
10 Jan 04:56:56 ntpd[939]: leapsecond file ('/usr/share/zoneinfo/leap-seconds.list'): loaded, expire=2022-12-28T00:00:00Z last=2017-01-01T00:00:00Z ofs=37
10 Jan 04:56:56 ntpd[939]: Listen and drop on 0 v6wildcard [::]:123
10 Jan 04:56:56 ntpd[939]: Listen and drop on 1 v4wildcard 0.0.0.0:123
10 Jan 04:56:56 ntpd[939]: Listen normally on 2 lo 127.0.0.1:123
10 Jan 04:56:56 ntpd[939]: Listen normally on 3 eth0 192.168.1.130:123
10 Jan 04:56:56 ntpd[939]: Listen normally on 4 lo [::1]:123
10 Jan 04:56:56 ntpd[939]: Listen normally on 5 eth0 [fe80::e65f:1ff:fe68:6a1e%2]:123
10 Jan 04:56:56 ntpd[939]: Listening on routing socket on fd #22 for interface updates
10 Jan 04:56:57 ntpd[939]: Soliciting pool server 74.6.168.73
10 Jan 04:56:58 ntpd[939]: Soliciting pool server 178.62.250.107
10 Jan 04:56:59 ntpd[939]: Soliciting pool server 93.94.88.51
10 Jan 04:57:00 ntpd[939]: Soliciting pool server 69.164.202.100
10 Jan 04:57:01 ntpd[939]: Soliciting pool server 91.189.89.199
10 Jan 04:58:02 ntpd[939]: Soliciting pool server 23.131.160.7
10 Jan 04:58:03 ntpd[939]: Soliciting pool server 83.151.207.133
10 Jan 04:58:05 ntpd[939]: Soliciting pool server 121.174.142.81
10 Jan 04:58:07 ntpd[939]: Soliciting pool server 216.229.4.66
10 Jan 04:58:08 ntpd[939]: Soliciting pool server 91.189.94.4
10 Jan 04:59:06 ntpd[939]: Soliciting pool server 23.157.160.168
10 Jan 04:59:08 ntpd[939]: Soliciting pool server 217.91.44.17
10 Jan 04:59:09 ntpd[939]: Soliciting pool server 46.249.42.12
10 Jan 04:59:11 ntpd[939]: Soliciting pool server 104.194.8.227
10 Jan 04:59:14 ntpd[939]: Soliciting pool server 91.189.89.198
10 Jan 05:00:13 ntpd[939]: Soliciting pool server 202.28.116.236
10 Jan 05:00:13 ntpd[939]: Soliciting pool server 66.228.48.38
10 Jan 05:00:15 ntpd[939]: Soliciting pool server 202.65.114.202
10 Jan 05:00:17 ntpd[939]: Soliciting pool server 72.30.35.88
10 Jan 05:00:21 ntpd[939]: Soliciting pool server 91.189.91.157
```

Not unique to Adafruit (I'm not using anything from Adafruit on this), so may post elsewhere.