adafruit/Raspberry-Pi-Installer-Scripts #98: /dev/fb1 Not Accessible on Pi4

mattburch, 2020-07-09:
Running the 'adafruit-pitft.sh' script provides that ability to mirror /dev/fb0 framebuffer but /dev/fb1 is never displayed on the Pi4.

ladyada, 2020-08-03:
hi we made some changes to the script recently, including some changes to framebuffer creation - please try again with latest commits!

closed 2020-08-03 by ladyada

mattburch, 2020-08-04:
Hello,

Executing that latest pull of the repo still produces a single /dev
reference to /dev/fb0 without a buffer for /dev/fb1.

ladyada, 2020-08-04:
we need more info on your setup to reproduce

mattburch, 2020-08-05:
I have the 2.2" PiTFT no touch Adafruit HAT installed on a Pi4 with the
following build details:

```
$ uname -a
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020
armv7l GNU/Linux
```

```
$ cat /etc/debian_version
10.4
```

```
$ lsmod | grep fb
fb_ili9340             16384  1
fbtft                  45056  1 fb_ili9340
syscopyarea            16384  2 fbtft,drm_kms_helper
sysfillrect            16384  2 fbtft,drm_kms_helper
sysimgblt              16384  2 fbtft,drm_kms_helper
fb_sys_fops            16384  2 fbtft,drm_kms_helper
```

I have followed the Adafruit instructions for setting up the HAT from here
<https://learn.adafruit.com/adafruit-2-2-pitft-hat-320-240-primary-display-for-raspberry-pi/easy-install>
and
have configured the display for no console output and to not mirror the
HDMI interface. My goal is to get two FB* /dev interfaces that can be
leveraged from manual redirection and maintain a TTY console. But the
install only appears to establish the default /dev/fb0 display.

```
$ ls /dev | grep fb
fb0
```

ladyada, 2020-08-06:
you may want to 'force' HDMI output on
https://raspberrypi.stackexchange.com/questions/2169/how-do-i-force-the-raspberry-pi-to-turn-on-hdmi

mattburch, 2020-08-08:
Thank you for the information, this appears to have addressed my issues.