adafruit/Raspberry-Pi-Installer-Scripts #100: Raspbian Buster: Apt failed to compile ST7789V driver

chardane, 2020-07-23:
## Summary
Hello! I have a Pi Zero on Raspberry Pi Buster and trying to run `./adafruit-pitft.sh` following the instructions in the [Adafruit Mini PiTFT - Color TFT Add-ons for Raspberry Pi Guide](https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/kernel-module-install). But I am getting the error `WARNING: Apt failed to compile ST7789V driver!`. 

The full error stack trace is [in this gist](https://gist.github.com/chardane/d60176048d6b94735159baa406f62a2a), but I believe the relevant stack trace is this:
```
make: Entering directory '/usr/src/linux-headers-5.4.51+'
  CC [M]  /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.o
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c:24:10: fatal error: drm/tinydrm/mipi-dbi.h: No such file or directory
 #include <drm/tinydrm/mipi-dbi.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.o] Error 1
make: *** [Makefile:1709: /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.51+'
WARNING : Apt failed to compile ST7789V driver!
```

I've been able to make the script and the Mini PiTFT work by using **Raspbian Stretch** instead of **Buster**, but I'm hoping to be able to make it work on Buster as well since I need Python 3.6+ (I'm currently trying to update Python 3.5 -> 3.6 on Stretch but it's not going so well -- `pyenv install` seems to fail and compiling from source is taking a few hours as you can imagine)

## Things I've tried 

* I've checked to see if the folder specified in the `include` statement exists, but it looks like it does not:
```
pi@raspberrypi:~/Raspberry-Pi-Installer-Scripts $ ls /usr/src/linux-headers-5.4.51+/include/drm/tinydrm
ls: cannot access '/usr/src/linux-headers-5.4.51+/include/drm/tinydrm': No such file or directory
```

* I've also spotted [this issue](https://github.com/notro/tinydrm/issues/17) which seems to be related maybe, and suggests that this is a known issue, but I don't know enough about how to add or compile linux kernels, so I'm a bit out of my depth in that thread :)

## More Context
Here are the steps that I took, and the specific stack trace, I'm hoping someone can assist and let me know if i'm doing something wrong.

>Raspberry Pi OS: Buster (Linux raspberrypi 5.4.51+)
>Model: Pi Zero W
>PiTFT display: [Adafruit Mini PiTFT 1.14" 240x135](https://www.adafruit.com/product/4393)

1. Burned an SD card using the [Raspberry Pi Imager](https://www.raspberrypi.org/blog/raspberry-pi-imager-imaging-utility/). Running `uname -a` yields:
```
$ uname -a
Linux raspberrypi 5.4.51+ #1326 Fri Jul 17 10:39:45 BST 2020 armv6l GNU/Linux
```
2. Following instructions from the guide mentioned above, ran these to update:
```
sudo apt update -y
sudo apt-get update -y
sudo apt-get upgrade -y
```
3. Then, I shutdown with `sudo shutdown -h now`, attached the PiTFT 1.14, cloned this repo and ran the script like so:
```
cd ~
git clone https://github.com/adafruit/Raspberry-Pi-Installer-Scripts.git
cd Raspberry-Pi-Installer-Scripts
chmod +x adafruit-pitft.sh
sudo ./adafruit-pitft.sh
```
4. I selected `7. MiniPiTFT 1.14" display (240x135) - WARNING! CUTTING EDGE! WILL UPGRADE YOUR KERNEL TO LATEST` and then `1` for orientation.

The full error stack trace is [in this gist](https://gist.github.com/chardane/d60176048d6b94735159baa406f62a2a), but I believe the relevant stack trace is this:
```
make: Entering directory '/usr/src/linux-headers-5.4.51+'
  CC [M]  /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.o
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c:24:10: fatal error: drm/tinydrm/mipi-dbi.h: No such file or directory
 #include <drm/tinydrm/mipi-dbi.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.o] Error 1
make: *** [Makefile:1709: /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.51+'
WARNING : Apt failed to compile ST7789V driver!
```

I hope someone can help point me in the right direction, thank you so much!

ladyada, 2020-07-23:
did you try the latest piOS buster? just so we can reproduce the exact same setup

chardane, 2020-07-23:
Yes, I believe its the latest version of Buster (the one released on 2020-05-27)
```
$ uname -a
Linux raspberrypi 5.4.51+ #1326 Fri Jul 17 10:39:45 BST 2020 armv6l GNU/Linux
```
I hope that helps, let me know if there's another thing i can do to verify setup

lakai, 2020-07-27:
same issue here, pizerow , minipitft, newest raspbian.

ladyada, 2020-07-27:
the new kernel must have really changed stuff. you can either use the 'userspace' code or you can use an older release of piOS for now

chardane, 2020-07-27:
Thank you for looking into it! I will attempt to use Stretch and upgrade Python or I will use the userspace code. At least I now know it wasn't some missing dependency I failed to install 👍  Cheers!

ladyada, 2020-07-27:
im in the process of 'fixing' up the driver now. ill let you know when its ready. many things changed but i do have the basics running on the 5.4 kernel!

ladyada, 2020-07-28:
ok rewrote the driver, please remove the old clone of this repo and clone this branch instead
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/tree/st7789redux
then run script as usual. it should work? i just tried it on 5.4.51-v7l+

chardane, 2020-07-28:
Confirmed, works on the same setup described in the original post of this thread! Thank you so much! 🎉

ladyada, 2020-07-28:
ok ill merge it after we get one more test run going. closing this for now

closed 2020-07-28 by ladyada