adafruit/Raspberry-Pi-Installer-Scripts #102: Installation of st7789 display via adafruit-pitft.sh fails

OopsDoh, 2020-07-27:
I recently updated RaspberryPi OS (and tried a new installation as well) and now the installation of the [2.0'' 320x240 spi display](https://www.adafruit.com/product/4311) (st7789) with [adafruit-pitft.sh](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/adafruit-pitft.sh) fails (installation support [requested/suggested here](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/80) and added in [this pr](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/86).

I think the reason is that [fbtft_device and flexfb are gone in 5.4](https://github.com/notro/fbtft/wiki#fbtft_device-and-flexfb-are-gone-in-54)

In previous installations the st7789 displays have been enabled by adding fbtft_device init-sequence in /boot/config.txt

Unfortunaltely this method doesn't work anymore with a current RaspberryPi OS with Linux 5.4 kernel.

For testing I also tried to select the 2.2'' display in the installation script: The 2.0'' display is functioning but only shows an distorted image. No wonder because the 2.2 display is no st7789 driver. But when selecting this wrong display size, there is an dtoverlay added in the boot/config.txt

I guess there has to be added an [Device Tree overlay](https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays) for the st7789 displays. Unfortunately I didn't yet understand how this works.

I found that in the current [Raspberry-Linux 5.4](https://github.com/raspberrypi/linux/tree/rpi-5.4.y/arch/arm/boot/dts/overlays) there are already some overlays available for the [PiTFT2.2](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/pitft22-overlay.dts), [PiTFT 2.8 capacitive](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/pitft28-capacitive-overlay.dts), [PiTFT 2.8 resistive](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/pitft28-resistive-overlay.dts) and [PiTFT 3.5 resistive](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/pitft35-resistive-overlay.dts)

Could you/someone please help to create such an overlay for the st7789 displays and updating the PiTFT installer script? Or do you know if there is another solution for this issue?

Possibly related issue: [Raspbian Buster: Apt failed to compile ST7789V driver](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/100) (but that's about compiling and not using the pitft installer script)

ladyada, 2020-07-27:
exciting. we still have SCSI drivers from 1980 in the linux kernel but an actively used display driver gets ripped out :(
we'll take a look in the next few weeks. for now please try using either an older version of piOS or use 'userspace' for the display

ladyada, 2020-07-27:
@notro with `flex_fb` taken out, we will have to write/compile a `st7789v` driver right? there's no way to change the init commands dynamically anymore?

notro, 2020-07-27:
It's possible to override the init sequence with Device Tree. I just put together a wiki page since I couldn't find one (haven't done much fbtft stuff lately): https://github.com/notro/fbtft/wiki/Device-Tree

Many of the fbtft drivers are MIPI DBI compatible so they can be used interchangeably (those that use 0x2A, 0x2B, 0x2C for address window, doesn't set the ```->set_addr_win``` callback). The only thing that differ is init sequence, possibly gamma and rotation. Rotation is the tricky one since it overrides what's set in the ```init=``` property. So if the st7789v driver doesn't match your display, pick a driver that doesn't have a ```->set_var``` function.

ladyada, 2020-07-27:
yeah the window command for ST77xx and ILI are identical, as is 'display on/off' .... its just the init code and of course the display size that is different as its 240x240 not 240x320 - ill take a look at that page!
thanksfully we have userspace python drivers, and fb mirroring code, so folks can use the displays for just anything but console

ladyada, 2020-07-28:
@OopsDoh ok rewrote the driver, please remove the old clone of this repo and clone this branch instead
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/tree/st7789redux
then run script as usual. it should work? i just tried it on 5.4.51-v7l+

antonakv, 2020-07-28:
`This script downloads and installs
PiTFT Support using userspace touch
controls and a DTO for display drawing.
one of several configuration files.
Run time of up to 5 minutes. Reboot required!

Select configuration:
1. PiTFT 2.4", 2.8" or 3.2" resistive (240x320)
2. PiTFT 2.0" no touch (240x320)
3. PiTFT 2.2" no touch (240x320)
4. PiTFT 2.8" capacitive touch (240x320)
5. PiTFT 3.5" resistive touch (320x480)
6. PiTFT Mini 1.3" or 1.54" display (240x240)
7. MiniPiTFT 1.14" display (240x135) - WARNING! CUTTING EDGE! WILL UPGRADE YOUR KERNEL TO LATEST
8. Uninstall PiTFT
9. Quit without installing

SELECT 1-9: 7
Select rotation:
1. 90 degrees (landscape)
2. 180 degrees (portait)
3. 270 degrees (landscape)
4. 0 degrees (portait)

SELECT 1-4: 2
[PITFT] Checking init system...
Found systemd
/boot is mounted
[PITFT] System update
Updating apt indexes...
...
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

......
Reading package lists...
.........
[PITFT] Installing Python libraries & Software...
Installing Pre-requisite Software...This may take a few minutes!
[PITFT] Updating /boot/config.txt...
/boot/overlays/drm-minipitft114.dtbo: Warning (unit_address_vs_reg): /fragment@0/__overlay__/spidev@0: node has a unit name, but no reg property
############# UPGRADING KERNEL ###############
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
make: Entering directory '/usr/src/linux-headers-5.4.51+'
  CC [M]  /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.o
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c:24:10: fatal error: drm/tinydrm/mipi-dbi.h: No such file or directory
 #include <drm/tinydrm/mipi-dbi.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.o] Error 1
make: *** [Makefile:1709: /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.51+'
WARNING : Apt failed to compile ST7789V driver!
`

ladyada, 2020-07-28:
@antonakv you need to be on the branch `st7789vredux` not on `master` - please check? :)

antonakv, 2020-07-28:
``` $ git clone https://github.com/adafruit/Raspberry-Pi-Installer-Scripts.git
Cloning into 'Raspberry-Pi-Installer-Scripts'...
remote: Enumerating objects: 43, done.
remote: Counting objects: 100% (43/43), done.
remote: Compressing objects: 100% (32/32), done.
remote: Total 416 (delta 22), reused 26 (delta 11), pack-reused 373
Receiving objects: 100% (416/416), 143.60 KiB | 528.00 KiB/s, done.
Resolving deltas: 100% (231/231), done.
$ cd Raspberry-Pi-Installer-Scripts/
$ git checkout st7789vredux
error: pathspec 'st7789vredux' did not match any file(s) known to git
$ cd ..
$ git clone https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/tree/st7789redux
Cloning into 'st7789redux'...
fatal: repository 'https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/tree/st7789redux/' not found ```

ladyada, 2020-07-28:
check https://devconnected.com/how-to-clone-a-git-repository/ for a tutorial on cloning branches

antonakv, 2020-07-28:
It works with branch st7789vredux. Thanks @ladyada .

OopsDoh, 2020-07-28:
Thank you @ladyada - I only have the 320x240 2.0'' displays here for testing, not the 240x240 or smaler one.

It's not working in my tests with the 320x240 display.

In your branch the file **minipitft20-overlay.dts** is missing and producing an error.

I copied minipitft114 and I am currently trying to get it working. So far I get a console shown on the display, but only in a 240x240 square.

I guess there is anywhere defined to cut of the rest off (as not required when using the 240x240 display), I'm still searching for a way to change it, but it's taking some time...

As far as I know those st7789 displays are vertically defined, so 240 is always the width and the height can change like from 240 to 320. Maybe there has to be changed someting in the **st7789v_ada.c** file but at my first attempt it seems I crashed that file when saving probably in a wrong file format. I'll keep on testing...

ladyada, 2020-07-28:
@OopsDoh are you wiring it by hand? i dont recall we had a pitft that used that screen. i have to make a DTO for a 320x240 display

OopsDoh, 2020-07-28:
I use a simple text editor to make changes in the files and then execute the script again - did you mean that with "writing it by hand"? I'm not familiar with C++ coding editors, just trial and error...

Yes I know that [this 2.0'' display](https://www.adafruit.com/product/4311) isn't officially named "PiTFT" but it was working perfectly with the PiTFT installer script after the intergration by @makermelissa [here](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/86) - before the RaspiOS used Linux 5.4 kernel... It's basically the same driver as the 240 and smaler st7789 displays are using.

By the way you can change the spi-max-frequency to 64000000 - that was working well before, at least with the 320x240 display (and should also with the 240x240 display).

ladyada, 2020-07-28:
ok! i just committed a dts - please try it!

ladyada, 2020-07-28:
updated script as well, put 2.0" at the bottom

OopsDoh, 2020-07-28:
Tests are running... first attempt looks good (full width console) - I'll check all variants (console/HDMI mirror and rotations)

OopsDoh, 2020-07-29:
Here some updates: Unfortunatelly the 2'' 320x240 is not working correctly yet. I made a test series and tried different options and changes in the overlay file, but haven't found the correct settings yet.

Here some findings that should be changed in the [adafruit-pitft.sh](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/st7789redux/adafruit-pitft.sh) file:

Following lines can be deleted (clean up), these variables aren't used anymore (because there are no init sequences anymore):
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/d046e3fd424102174592004a12c8f0b340b70089/adafruit-pitft.sh#L55-L63

The IF-ELSE in this block should be reverted:
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/d046e3fd424102174592004a12c8f0b340b70089/adafruit-pitft.sh#L430-L437
The IF statement was added as a rotation correction for the 320x240-display in the old implementation (with fbtft_device), I think in the current version only those two lines are required:
```
WIDTH=`python -c "print(int(${WIDTH_VALUES[PITFT_SELECT-1]} * ${SCALE}))"`
HEIGHT=`python -c "print(int(${HEIGHT_VALUES[PITFT_SELECT-1]} * ${SCALE}))"`
```
Also important are changes in this block, this is currently wrong:
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/d046e3fd424102174592004a12c8f0b340b70089/adafruit-pitft.sh#L571-L575

The ordner of the WIDTH and HEIGHT values is related to the order in PITFT_TYPES, because those values are used for the size of the HDMI monitor resolution. So it should be changed like this:

```
PITFT_TYPES=("28r" "22" "28c" "35r" "st7789_240x240" "st7789_240x135" "st7789_240x320")
WIDTH_VALUES=(320 320 320 480 240 240 320)
HEIGHT_VALUES=(240 240 240 320 240 135 240)
HZ_VALUES=(64000000 64000000 64000000 32000000 64000000 64000000 64000000)
```

I think the HZ_VALUES are currently not used anywhere else in the script, so this line can be deleted as well?

Another finding: If I understand it correctly, you can define "overwrites" to the default values defined in the overlay files.
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/d046e3fd424102174592004a12c8f0b340b70089/overlays/minipitft13-overlay.dts#L70-L77
So it should be possible to change this block:
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/d046e3fd424102174592004a12c8f0b340b70089/adafruit-pitft.sh#L253-L256
to something like this:
```
if  [ "${pitfttype}" == "st7789_240x320" ]; then
    dtc -@ -I dts -O dtb -o /boot/overlays/drm-minipitft13.dtbo overlays/minipitft13-overlay.dts
    overlay="dtoverlay=drm-minipitft13,rotation=${pitftrot},height=240,width=320,col_offset=0,row_offset=0"
fi
```
and this should have the same result as when you're using **st7789v_240x320-overlay.dts** - is this correct?

At the moment the 320x240 screen doesn't work when I change either height or width to 320.

When I keep 240x240 I get results like this:

[image]

A bit strange is that sometimes during my tests (just for a second after the booting sequence) the content on the display is shown in full size (320x240) and then switching back to a cropped square view with distorded border.

@ladyada What are the 3rd and 4th value in this file ("58, 43")?
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/d046e3fd424102174592004a12c8f0b340b70089/st7789_module/st7789v_ada.c#L129

I'll keep on testing and trying to find a way to get the display running correctly again... it's very time consuming unfortunately

OopsDoh, 2020-07-29:
I got the console working in all rotation modes (and full width/height) when both height and width are switched here once again:
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/a3ea15cfd94920429a73249c65a24ecf99222855/adafruit-pitft.sh#L563-L564
to this (values at the end):
```
WIDTH_VALUES=(320 320 320 480 240 240 240)
HEIGHT_VALUES=(240 240 240 320 240 135 320)
```
Probably required because it's a portrait display natively.

But I'm still struggling with the HDMI mirroring.

I think the tricky part is to redefine the rotation for this display and/or the fbcp because of the swaped height/width.

But currently I found/have the problem, that the UPDATE KERNELL function seems to be called twice and I always get `FATAL ERROR: Couldn't open "/overlays/st7789v_240x320-overlay.dts": No such file or directory` at the end.

The "missing" file is not missing, it's still in the cloned folder.

Also I'm not sure if it's wise to rerun the script or if I always have to start with a new clean RaspiOS installation...

OopsDoh, 2020-07-29:
I think this can be removed as well, because flexfb and fbtft_device are gone...
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/a3ea15cfd94920429a73249c65a24ecf99222855/adafruit-pitft.sh#L213-L219
and this line
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/a3ea15cfd94920429a73249c65a24ecf99222855/adafruit-pitft.sh#L224

OopsDoh, 2020-07-30:
I'm stuck. Console output is working correctly. But HDMI mirror is only showing a black sceen no matter what parameters I change.

Here is my current version of a modified **adafruit-pitft.sh**:

https://github.com/OopsDoh/Raspberry-Pi-Installer-Scripts/blob/patch-2/adafruit-pitft.sh

When selecting HDMI mirror the function **update_configtxt()** was called multiple times (because of the rotation hacks) and this was also executing the update kernel function multiple times.

To avoid this I added a new function **st7789_overlay_install()** to have this part outside the **update_configtxt()**.

I also added a variable **SCRIPT_FOLDER_PATH** to avoid the error about missing overlay file.

I'm wondering if HDMI mirroring is really working with the smaler 240x240 and 240x135 display and your version of the install script.

Can you please test HDMI mirroring with those displays again (90°/180°/270°)? I don't understand how those could work but the 320x240 does not...

OopsDoh, 2020-08-02:
OK I'll wait... In the meantime for others who need to mirror the HDMI display to the SPI display:

Start with a RaspberryPi OS with 4.9 kernel.

To prevent getting the kernel upgrad to 5.4 and keep 4.9 enter this in console:

```
sudo apt-mark hold libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0
sudo apt-mark hold raspberrypi-bootloader raspberrypi-kernel raspberrypi-kernel-headers
```
(code found [here](https://www.raspberrypi.org/forums/viewtopic.php?t=281519&p=1705079#p1705079), originally from [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1701547#p1701547))

This holds your system on 4.9

Unfortunatelly in my test with the current version of the adafruit-pitft.sh the "Update Kernel" functions gets stuck in a loop (stopping with message "Please reboot and run the script again")

So just use the previous version of the adafruit-pitft.sh available [here](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/tree/6336cf1a730757264e91ea0c7bade7a00cd98b8c).

@ladyada : I think HDMI mirror via fbcp won't possible with the new kernel at all. I had a look at [the code of rpi-fbcp](https://github.com/adafruit/rpi-fbcp/blob/master/main.c) and searched for some used functions, eg "DISPMANX". In [this forum post](https://www.raspberrypi.org/forums/viewtopic.php?t=269769#p1636551) there's a comment _"Note: kms won’t ever support applications that use firmware apis for display like **dispmanx**, or mmal/openmax video_render, including omxplayer, raspivid and non-gbm kodi. Similarly tvservice is not useful in this environment."_ So there is probably another solution with "KMS" required (whatever kms is) to mirror the HDMI fo the SPI framebuffer...

ladyada, 2020-08-03:
@OopsDoh you can try this PR (you will need to switch to the branch) to try a new variation on the driver which uses the older fbtft structure
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/104/files

OopsDoh, 2020-08-03:
Thank you!

Here are my results (tested on new RaspiOS installation, updatet to 5.4:

left column: first test round with HDMI mirror,
right column: second test round with console

[image]

To correct the compressed views (my tests no 2 and 4) this block has to be changed:
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/8bc1771bb2b1e40202058f39cc136620ffecfb39/adafruit-pitft.sh#L424-L431
to just those two lines:
```
WIDTH=`python -c "print(int(${WIDTH_VALUES[PITFT_SELECT-1]} * ${SCALE}))"`
HEIGHT=`python -c "print(int(${HEIGHT_VALUES[PITFT_SELECT-1]} * ${SCALE}))"`
```
After these changes everything seems to be working correctly:

[image]

Thank you once again, that's awesome!

ladyada, 2020-08-03:
ok great that is very helpful!

ladyada, 2020-08-03:
closing as resolved (for now!)

@notro something i noticed that threw me off...  when moving a TFT display driver from the older 'fbtft' driver style
	https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/staging/fbtft/
to
	https://github.com/raspberrypi/linux/tree/rpi-5.4.y/drivers/gpu/drm/tiny

if there is NO other framebuffer device (/dev/fbX) such as an HDMI or TV monitor plugged into the pi
there is only a single /dev/fb0 that belongs to the TFT display.
the TFT display works fine with either fbtft or tiny DRM, is initialized and can be used as a framebuffer.
with tiny DRM, the TFT /dev/fb0 will automatically get the text console.

However, if there is ANY other framebuffer device (/dev/fbX) such as an HDMI or TV monitor plugged into the pi
so that there are multiple framebuffer devices, /dev/fb0 will be the HDMI/TV display and /dev/fb1 will
belongs to the TFT display.
with tiny DRM, the TFT /dev/fb1, the kernel module is loaded, gets probed
	e.g. https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/gpu/drm/tiny/mi0283qt.c#L180
but the enable function is never called
        e.g. https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/gpu/drm/tiny/mi0283qt.c#L50
so the display is never initialized, and no SPI traffic is seen.
The /dev/fb1 device is definitely made, but its never 'enabled' ... i think this might be a bug? or maybe im missing something. should i post this to a kernel mailing list?

closed 2020-08-03 by ladyada

notro, 2020-08-03:
Yeah that one can be confusing, from the tinydrm [wiki](https://github.com/notro/tinydrm/wiki#fbtftfbdev-vs-drm):

```fbtft turns on the display when probed - DRM turns on the display when first used.```

fbcon (as configured on the Pi) automatically puts a console on the first framebuffer ie. /dev/fb0. If it was mi0283qt that created it, then the display is enabled and a console is shown on the TFT. However if the display gets /dev/fb1 then there is no console attached automatically so there's no user of the DRM device. From the fbdev side, attaching a console ```con2fb 1 1``` or opening /dev/fb1 followed by a FBIOPUT_VSCREENINFO ioctl will enable the display.

[modetest](https://github.com/notro/tinydrm/wiki/Development#modetest) is a nice tool for testing from the DRM side of things.

ladyada, 2020-08-03:
ooh shoot i didnt see that wiki - would have saved me a few hours :)
ok i guess i could put the drm 'tickle' in a systemd service or something

lbendlin, 2020-08-04:
Tried the latest script on a Pi Zero W with a fresh Raspbian Lite image and the 240x240.  First run of the script updated the kernel, second run of the script aborted thusly:

[PITFT] Updating /boot/config.txt...
FATAL ERROR: Couldn't open "overlays/minipitft13-overlay.dts": No such file or directory
############# UPGRADING KERNEL ###############
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
./adafruit-pitft.sh: line 265: pushd: st7789_module: No such file or directory
make: Entering directory '/usr/src/linux-headers-5.4.51+'
scripts/Makefile.build:42: /home/pi/Makefile: No such file or directory
make[1]: *** No rule to make target '/home/pi/Makefile'.  Stop.
make: *** [Makefile:1709: /home/pi] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.51+'
WARNING : Apt failed to compile ST7789V drivers!

Do I still need to use the script from the branch?

makermelissa, 2020-08-04:
@lbendlin make sure you are cloning the whole repo. Instructions are at https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/1-3-240x240-kernel-module-install.

lbendlin, 2020-08-04:
@makermelissa sorry about that.

I get a warning 
/boot/overlays/drm-minipitft13.dtbo: Warning (unit_address_vs_reg): /fragment@0/__overlay__/spidev@0: node has a unit name, but no reg property
There is a cute little typo in the script at the rotation selection. Says "(portait)".  Rotation settings are not working for any of the options. Either it stays in mode 1 or the screen is cut off on one of the sides. Do I have to uninstall before changing orientation?

The shell colors (grey/green) seem to be ignored, console comes up all white.

makermelissa, 2020-08-04:
You can ignore that warning. I just added the fix for the typos to an outstanding PR that I have, so thanks for catching that. No, you shouldn't need to uninstall first. It sounds like it might be a genuine issue that we'll need to address.

lbendlin, 2020-08-04:
might be some initialization issue. The screen starts working after a second reboot, and pygame works as expected.