adafruit/Raspberry-Pi-Installer-Scripts #105: Trouble with Portrait Mode with adafruit-pitft.sh

makermelissa, 2020-08-03:
When testing the latest changes, I noticed that portrait mode was not working on the display. After a bit of research, I found that commenting out `dtoverlay=vc4-kms-v3c` in `/boot/config.txt` fixes the issue. I'm not sure if we want to update the `adafruit-pitft.sh` to comment out or remove this line. Tested on ILI9341, but I also noticed the issue on the ST7789. 

The only thing is doing this appears to make touch input stop working though, so perhaps this is not the best solution. I'll have to look into this more.

makermelissa, 2020-08-03:
One other thing I forgot to mention is with the overlay enabled, with manually going through the GUI and changing the orientation, everything works fine.

makermelissa, 2020-08-03:
Changing the rotate parameter in the dtoverlay for the LCD works with the vc4-fkms-v3d enabled, though it is not rotated on the HDMI display, but it appears squished (unless the width and height in hdmi_cvt are swapped), so that's another option.

makermelissa, 2020-08-03:
Oh, bu then HDMI stops working because it doesn't like the dimensions.

makermelissa, 2020-08-03:
Re-ran the `adafruit-pitft.sh` (in it's latest incarnation) with `dtoverlay=vc4-fkms-v3d` commented out and touch is working along with the rotation appearing correct. So it looks like the recommended fix is commenting out `dtoverlay=vc4-fkms-v3d`.

@ladyada, I can modify the script so it checks for this and comments it out (and uncomments on an uninstall) if you think that would be the right approach.

ladyada, 2020-08-03:
i think thats ok - go for it!

PR #106 (2020-08-03): Fix for Pi 4 portrait rotation
Fixes #105.
closed 2020-08-03 by makermelissa