adafruit/Raspberry-Pi-Installer-Scripts #108: modprobe: ERROR: could not insert 'snd_i2s_rpi': Exec format error

bablokb, 2020-08-08:
I tried to install the i2s-mic kernel module on a pi-zero (PiOS Buster-Lite 2020-05). The image  uses a 4.19 kernel, but the foundation has updated the raspberry-pi-kernel-headers package to 5.4, so the install fails. After running rpi-update, I am running a 5.4 kernel and the i2smic.sh script works fine (the modules build without problems). But trying to load the module fails with the given error.

ladyada, 2020-08-08:
yah you need the latest kernel for sure - a *lot* changed from 4 to 5 @caternuson want to add a check/warning?

caternuson, 2020-08-08:
@ladyada Still works per guide. Added a general warning about making sure to follow all steps in order shown. Doing things out of sequence can definitely lead to building against the wrong kernel headers.

@bablokb Don't use `rpi-update`. Update your system using the `apt-get update` / `upgrade` combo. I just ran through the guide and it should work if you follow the process there:
https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/raspberry-pi-wiring-test
Make sure to also reboot where indicated.

I started with the `2020-05-27-raspios-buster-lite-armhf` image, which is based on the 4.19 kernel, so immediately after installing get:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
```
After running `sudo apt-get update` / `upgrade`:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
```
No change, that's because a reboot is needed. After **reboot**:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.51+ #1327 Thu Jul 23 10:53:06 BST 2020 armv6l GNU/Linux
```
Now showing updated kernel. Now we can run the I2S mic install script. After running script **AND rebooting**:
```
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: sndrpii2scard [snd_rpi_i2s_card], device 0: simple-card_codec_link snd-soc-dummy-dai-0 [simple-card_codec_link snd-soc-dummy-dai-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
All good.

closed 2020-08-08 by ladyada