adafruit/Raspberry-Pi-Installer-Scripts #116: miniPiTFT 1.3" 240x240 doesn't work: black screen only

christastic, 2020-09-27:
I installed kernel module as described in
https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/1-3-240x240-kernel-module-install

`
This script downloads and installs
PiTFT Support using userspace touch
controls and a DTO for display drawing.
one of several configuration files.
Run time of up to 5 minutes. Reboot required!

Select configuration:
1. PiTFT 2.4", 2.8" or 3.2" resistive (240x320)
2. PiTFT 2.2" no touch (240x320)
3. PiTFT 2.8" capacitive touch (240x320)
4. PiTFT 3.5" resistive touch (320x480)
5. PiTFT Mini 1.3" or 1.54" display (240x240) - WARNING! WILL UPGRADE YOUR KERNEL TO LATEST
6. MiniPiTFT 1.14" display (240x135) - WARNING! WILL UPGRADE YOUR KERNEL TO LATEST
7. ST7789V 2.0" no touch (240x320) - WARNING! WILL UPGRADE YOUR KERNEL TO LATEST
8. Uninstall PiTFT
9. Quit without installing

SELECT 1-9: 5
Select rotation:
1. 90 degrees (landscape)
2. 180 degrees (portrait)
3. 270 degrees (landscape)
4. 0 degrees (portrait)

SELECT 1-4: 1
[PITFT] Checking init system...
Found systemd
/boot is mounted
[PITFT] System update
Updating apt indexes...
...
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

......
Reading package lists...
.........
[PITFT] Installing Python libraries & Software...
Installing Pre-requisite Software...This may take a few minutes!
[PITFT] Updating /boot/config.txt...
Already have an adafruit-pitft-helper section in /boot/config.txt.
Removing old section...
/boot/overlays/drm-minipitft13.dtbo: Warning (unit_address_vs_reg): /fragment@0/__overlay__/spidev@0: node has a unit name, but no reg property
/boot/overlays/drm-minipitft13.dtbo: Warning (unit_address_vs_reg): /fragment@0/__overlay__/spidev@1: node has a unit name, but no reg property
############# UPGRADING KERNEL ###############
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module /home/pi/Raspberry-Pi-Installer-Scripts
make: Entering directory '/usr/src/linux-headers-5.4.51+'
  Building modules, stage 2.
  MODPOST 2 modules
  LD [M]  /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/fb_st7789v.ko
  LD [M]  /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.ko
make: Leaving directory '/usr/src/linux-headers-5.4.51+'
/home/pi/Raspberry-Pi-Installer-Scripts
Would you like the console to appear on the PiTFT display? [y/n] n
[PITFT] Making sure console doesn't use PiTFT
Removing console fbcon map from /boot/cmdline.txt
Screen blanking time reset to 10 minutes
Would you like the HDMI display to mirror to the PiTFT display? [y/n] y
[PITFT] Adding FBCP support...
Installing cmake...
Downloading rpi-fbcp...
/tmp /home/pi/Raspberry-Pi-Installer-Scripts
Uncompressing rpi-fbcp...
Building rpi-fbcp...
Installing rpi-fbcp...
/home/pi/Raspberry-Pi-Installer-Scripts
Remove fbcp from /etc/rc.local, if it's there...
We have systemd, so install fbcp systemd unit...
Created symlink /etc/systemd/system/multi-user.target.wants/fbcp.service → /etc/systemd/system/fbcp.service.
Configuring boot/config.txt for forced HDMI
Using native resolution
[PITFT] Success!

Settings take effect on next boot.

REBOOT NOW? [y/N] y
`

I tried with and without console, but the screen remains blank. I see the backlight is working but nothing will be displayed.

ladyada, 2020-10-19:
hi we made many updates lately, can you try the newest python installer script according to the guide? we just tested it :)

makermelissa, 2021-01-26:
Closing due to lack of response.

closed 2021-01-26 by makermelissa