adafruit/Raspberry-Pi-Installer-Scripts #128: Some settings not working with 1.3" TFT

caternuson, 2020-11-11:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=50&t=171623

Recreated with Pi Zero W, `2020-08-20-raspios-buster-armhf-lite.img`, and following Python script process per guide:
https://learn.adafruit.com/adafruit-1-3-color-tft-bonnet-for-raspberry-pi/kernel-module-install

Using guide example:
```
sudo python3 adafruit-pitft.py --display=st7789_240x240 --rotation=180 --install-type=console
```
results in:
[image]

Re-running and selecting 0 degree rotation results in:
[image]

caternuson, 2020-11-14:
Related?
https://forums.adafruit.com/viewtopic.php?f=8&t=170661#p834732

ladyada, 2020-11-14:
probably a typo - @makermelissa plz look on monday, something just got swapped around?

makermelissa, 2020-11-16:
Ok, it took forever to update my Pi, but I was able to reproduce as well.

makermelissa, 2020-11-16:
I'm getting the same issue when running adafruit-pitft.sh and choosing the same options. It could be something changes in the 8/20 release of Raspberry Pi that I missed. I'm going to look a little deeper into this.

closed 2020-11-16 by makermelissa

makermelissa, 2020-11-16:
I just updated the guide so it appears the correct direction too. Apparently it is 0 degrees if using the console and 180 if using FBCP.

PR #129 (2020-11-16): Fix 80 row offset issue for 1.3" display
Fixes #128. It looks like something changed with driver and it was not working in rotations other than 0. I used the driver for the 1.14" and adjusted the backlight pin. Tested with MiniPiTFT 1.3", PiTFT Bonnet 1.3", and BrainCraft HAT. Tested with various rotations in console and HDMI mirroring with successful results. Also fixed an issue where setting rotation to 0 ended up with 90 degree rotation due to a typo.