adafruit/Raspberry-Pi-Installer-Scripts #133: WARNING : Apt failed to install software! adafruit-pitft.sh

Collbrothers, 2020-12-26:
While running [this](https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/adafruit-pitft.sh) on my raspberry pi 4 with newly updated python and pip3.
When installing Pre-requisite Software it spits this out:
```
E: Unable to locate package python-pip
E: Package 'python-smbus' has no installation candidate
E: Unable to locate package python-spidev
```

ladyada, 2020-12-26:
somtheing is seriously messed up with your raspbian install.
make sure you have internet and update/upgrade your board. 
we only support raspbian/raspiOS

Collbrothers, 2020-12-27:
> we only support raspbian/raspiOS
Thaat makes sense, is there a future where Ubuntu server is supported?

ladyada, 2020-12-27:
if someone could submit a PR for the differences we'd look to add it, but its hard enough supporting raspbian we do not have plans to do it ourselves :)
closing for now

closed 2020-12-27 by ladyada