adafruit/Raspberry-Pi-Installer-Scripts #134: FBCP no longer seems to work with Adafruit 2.2" PiTFT HAT - 320x240 Display and Raspian OS

rolohaun, 2021-01-04:
This was working a week ago and suddenly no longer works on my raspberry pi zero W. I get the error below

Installing rpi-fbcp...
Remove fbcp from /etc/rc.local, if it's there...
We have systemd, so install fbcp systemd unit...
Setting raspi-config to boot to desktop w/o login...
Configuring boot/config.txt for forced HDMI
PITFT Updating X11 default calibration...
Traceback (most recent call last):
  File "adafruit-pitft.py", line 647, in <module>
    main()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "adafruit-pitft.py", line 637, in main
    if not update_xorg():
  File "adafruit-pitft.py", line 448, in update_xorg
    transform = "Option \"TransformationMatrix\" \"{}\"".format(pitft_config["touchscreen"]["transforms"][pitftrot])
TypeError: 'bool' object is not subscriptable

makermelissa, 2021-01-04:
Interesting. This appears to be caused by a click update. I'll go through and test it with the latest version.

aardvarkyvark, 2021-01-04:
HI, I also have the a "failed to disable unit: unit file fbcp.service does not exist" while installing on a 240x240. Not sure if it's a different issue or related.

makermelissa, 2021-01-05:
@aardvarkyvark it automatically tries to remove any existing fbcp services on install so there aren't duplicates or old ones. That just means it didn't find an existing one to remove.

aardvarkyvark, 2021-01-05:
Oh! Ok, thanks. I was able to install it and get it working, but the HDMI out was not displaying. Is there a features list or config doc somewhere that I can review for the basic options and functions when installing the display? Or is it just meant to be a lone mini-display without any other visual interface options?

makermelissa, 2021-01-05:
@rolohaun can you type `pip3 show click`? I want to see what version of Click you're using. Also are you installing Raspberry Pi OS Full or Lite? What command are you using to install it? I just tried with the latest version of Click (though 7.0.0 should work as well) and didn't receive that error on a Pi 4, but want to closer match your setup to reproduce the error.

@aardvarkyvark I'm not sure if you're using lite or full or how you're installing it.

rolohaun, 2021-01-05:
@makermelissa here is what I get
Name: Click
Version: 7.0
Summary: Composable command line interface toolkit
Home-page: https://palletsprojects.com/p/click/
Author: Armin Ronacher
Author-email: armin.ronacher@active-4.com
License: BSD
Location: /usr/lib/python3/dist-packages
Requires:
Required-by:

I am using Raspberry Pi OS with desktop
and using this command sudo python3 adafruit-pitft.py --display=22 --rotation=90 --install-type=fbcp from this page https://learn.adafruit.com/adafruit-2-2-pitft-hat-320-240-primary-display-for-raspberry-pi/easy-install

makermelissa, 2021-01-05:
Thanks, I'll adjust my setup a bit and try again.

makermelissa, 2021-01-05:
Ok! I was able to reproduce on a Pi 4 with Click 7.0 and a 2.2 screen using the above command. Hopefully I'll get this fixed soon.

rolohaun, 2021-01-05:
@makermelissa thanks for looking into it!

PR #136 (2021-01-05): Fixed issue with 2.2-inch PiTFT and touchscreen param
Fixes #134. This was supposed to be absent rather than False.
closed 2021-01-05 by makermelissa