adafruit/Raspberry-Pi-Installer-Scripts #135: 1.3" 240x240 Kernel Module Install removed HDMI output

aardvarkyvark, 2021-01-05:
Hello, thought I'd open up a new issue to cut down on the noise and compartmentalize. I want to use this display as a viewfinder for raspistill (ie. press a shutter and it displays the preview of the image until keypress).

I used this guide to install the minidisplay, and everything went fine in terms of the mini-display showing the console. I was able to run commands and updates, etc. https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/1-3-240x240-kernel-module-install

However when rebooting, plugging in HDMI monitor shows a black screen with a dash or maybe an underscore in the top left. The minidsiplay is still fully functional, but was hoping that HDMI would still output non-console info. Am I misunderstanding how this works? What is expected behaviour on completion of this guide?

aardvarkyvark, 2021-01-05:
I am using full pi OS (not lite), and installed as console.

makermelissa, 2021-01-05:
Yeah, I think it only displays something on both when running as FBCP.

protobruce, 2021-01-07:
Sorry to reopen - I'm not finding a 240x240 fbcp option. I've tried a few display options on the guide here but they're all 340x240: https://learn.adafruit.com/running-opengl-based-games-and-emulators-on-adafruit-pitft-displays?view=all
Is there one?

protobruce, 2021-01-07:
Update, I figured out the manual install using the pitft helper. I think I selected the 3.5 resistive with 0 rotation.
For reference, the bottom of my config.txt looks like this now:
```
# --- added by adafruit-pitft-helper Wed Jan  6 17:07:56 2021 ---
hdmi_force_hotplug=1
dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=off
dtoverlay=drm-minipitft13,rotation=0
# --- end adafruit-pitft-helper Wed Jan  6 17:07:56 2021 ---
dtoverlay=pitft35-resistive,rotate=0,speed=32000000,fps=60
display_rotate=0
hdmi_cvt=480 320 60 1 0 0 0
```
For my camera project, I'm still trying to get the raspistill --full preview to display on the 240x240 tft. With the monitor plugged in, I get the full preview on the HDMI output, but the console stays as-is. I tried adjusting hdmi hotplug to 0, but I can't seem to get the preview to drive into the tft. Any suggestions??

How would I force all vid output through the console screen if I needed to and ssh's the rest? Thanks for any assistance.

protobruce, 2021-01-07:
Found it in the Setting Up section here https://learn.adafruit.com/running-opengl-based-games-and-emulators-on-adafruit-pitft-displays?view=all using the 480x320 display name seems to be working so far. Will keep working and update here for reference.

makermelissa, 2021-01-08:
Thanks. Are you saying the 480x320 driver works as expected, but the 240x240 driver does not?

protobruce, 2021-01-08:
Sorry, no - the 480x320 FBCP driver does not work on this screen.

I've put in an issue over at the FBCP git for clarification on the CMAKE option https://github.com/juj/fbcp-ili9341/issues/183 because it lists the waveshare version command but not the adafruit. I have the driver number, but none of the options I'm using seem to work on the cmake line.

ladyada, 2021-01-08:
what is 'this screen' specifically?

protobruce, 2021-01-08:
Sorry just realized I didn't include it in the OP. It's the 240x240 https://www.adafruit.com/product/4484 mini pitft. The no-solder one that fits on the pins and has two buttons. I've been able to get it working with the AF guide on console as above, but want to try FBCP.

ladyada, 2021-01-08:
the way the pi works, it doesnt like having the output be 240x240 on hdmi so it doesnt appear on the hdmi output

protobruce, 2021-01-08:
Yes understood, I reopened because I wanted to figure out a way to drive the raspistill preview to the screen. I don't actually need HDMI output, I just need it to show the pi's video. Is there a way to do this via console?

Otherwise, I've been assuming I have to go the FBCP route and drive video output that way, so I'm looking for the correct 240x240 cmake option.

makermelissa, 2021-01-08:
Yes, to show raspistill on the display, fbcp is required. Even on lite.

protobruce, 2021-01-08:
If alright, I just want to leave this open until I get the correct CMAKE option line, and then I'll close it for future reference for anyone looking into getting this to work for this use case.

protobruce, 2021-01-11:
Hi @ladyada & @makermelissa I'm wondering if there is documentation on the 240x240 add-on display for pin wiring? It sounds like the ST7789 driver for fbcp-ili9341 should work but will need some manual adjustments in the CMAKE options line. Happy to plug away at it, just can't find anything on the technical details page. Failing that I'll do some trial and error and make some assumptions. https://www.adafruit.com/product/4484#technical-details

makermelissa, 2021-01-11:
Yes. https://learn.adafruit.com/adafruit-1-3-and-1-54-240-x-240-wide-angle-tft-lcd-displays/python-wiring-and-setup#st7789-and-st7735-based-displays-3042525-12

makermelissa, 2021-01-26:
Closing since there isn't much we can do on this.

closed 2021-01-26 by makermelissa