adafruit/Raspberry-Pi-Installer-Scripts #138: st7789_240x240 TFT freezes - raspberrypi-kernel incompatibility?

jepler, 2021-01-08:
Some recent update, possibly raspberrypi-kernel 1.20210104-1, has started causing my TFT to freeze.  Kernel has messages such as:
```
[   20.541654] fb_st7789v spi0.0: SPI transfer timed out
[   20.541683] spi_master spi0: failed to transfer one message from queue
[   20.541693] fb_st7789v spi0.0: fbtft_update_display: write_vmem failed to update display buffer
```
and
```
[   32.311669] WARNING: CPU: 2 PID: 508 at drivers/firmware/raspberrypi.c:63 rpi_firmware_transaction+0xec/0x128
[   32.311674] Firmware transaction timeout
...
[   32.311830] Workqueue: events get_values_poll [raspberrypi_hwmon]
...
[   32.312036] hwmon hwmon1: Failed to get throttled (-110)
[   33.351661] cpu cpu0: dev_pm_opp_set_rate: failed to find current OPP for freq 4294967186 (-34)
[   34.391658] raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-arm frequency: -110
```
(-34 is ERANGE and -110 is ETIMEDOUT)

makermelissa, 2021-01-08:
Hmm, yeah. I experienced this as well, so it's not isolated.

makermelissa, 2021-01-08:
Possibly related? https://github.com/raspberrypi/firmware/issues/1522

jepler, 2021-01-08:
When I switch back to an SD card with kernel 5.4.79-v7l+ (raspberrypi-kernel 1.20201201-1) it is fine, so it's not a hardware problem.

makermelissa, 2021-01-08:
I agree it's not hardware related. I was pointing out a PR that was related to the kernel that started on Jan 4.

jepler, 2021-01-08:
Pedro and I have also seen scrambled screens, like 
[image]

I _believe_ this is with the 20210104 kernel but I haven't actually had them verify that (that's Pedro's photo but I saw something similar)

makermelissa, 2021-01-08:
It looked similar on mine.

jepler, 2021-01-10:
https://github.com/raspberrypi/linux/issues/4047 looks rather similar, with these messages both appearing: 
 * "Firmware transaction timeout"
 * "cpu cpu0: dev_pm_opp_set_rate: failed to find current OPP for freq 4294967186 (-34)"

There's now a kernel 20210108 but the changelog does not provide any information about what was, er, changed.
```
lftp raspbian.raspberrypi.org:/raspbian/pool/main/r/raspbian-archive-keyring> exraspberrypi-firmware (1.20210108-1) buster; urgency=medium
  
  * firmware as of 0d458874a89921fbe460e422b239695e1e101e2b
  * Linux version 5.4.83

 -- Serge Schneider <serge@raspberrypi.com>  Fri, 08 Jan 2021 15:26:51 +0000
```
It wouldn't hurt to run the tft installer again to get the kernel update and see what happens.

jepler, 2021-01-10:
I upgraded my pi4 to raspberrypi-kernel 1.20210108-1, re-ran the installer script for the display, and my tft seems to be working now.  If someone else can confirm, we can treat this as a fixed glitch in the kernel that they quickly fixed -- if so, yay.

ladyada, 2021-01-10:
yayyy!

jepler, 2021-01-10:
.. something's wonky with sound now though.  (I didn't initially have speakers plugged in to my test rig).  I'll do some more testing and report back, but someone else's test would be nice too

jepler, 2021-01-10:
OK, ugh, it's a different issue for a different repo, but with a fresh full raspberry pi os install I went through the steps of installing the braincraft audio driver over kernel 1.20210108-1 and it's a staticy mess when I play lofi hip hop radio through it.  It's fine coming through HDMI (pactl set-default-sink 0) or the on board headphone jack (pactl set-default-sink 1) so not a generic problem, but an i2s problem.

makermelissa, 2021-01-12:
Ok, new issue opened at https://github.com/raspberrypi/firmware/issues/1524

Also, relevant forum thread: https://forums.adafruit.com/viewtopic.php?f=50&t=173947

makermelissa, 2021-01-12:
I have a new guide page to cover how to run the script to fix, but until it's published, for anyone watching this issue, here's a summary. 

1. Run these commands from a terminal:
```
cd ~
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/rpi-pin-kernel-firmware.sh
sudo sh rpi-pin-kernel-firmware.sh 1.20201126-1
```
2. Reboot
3. Re-run the adafruit-pitft.py installer script
4. Reboot again

makermelissa, 2021-02-25:
I think the root of the problem is that our `fbtft.h` file was a little out of date. I've been compiling against a fresh copy and have had no issues. Fixed up drivers coming soon.

closed 2021-02-26 by makermelissa

kcfino, 2021-11-09:
Is this issue solved? My son and I are barely following - we got to page 26 of the Machine Learning 101 and Braincraft PDF - and the hat stopped working. It only displays at shutdown and then it's inverted. I tried the above:
cd ~
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/rpi-pin-kernel-firmware.sh
sudo sh rpi-pin-kernel-firmware.sh 1.20201126-1
 
and then re-ran adafruit-pitft.py installer script - reboot.

But we are still no go on the TFT display... It starts lit up in back but then is dark. I've plugged in an external display and I only have a bluescreen now. I can still ssh though!

We are pretty frustrated at this point!!!

makermelissa, 2021-11-09:
Yes, the issue was solved. A new major release of Raspberry Pi just came out yesterday (Bullseye) and broke things. We're working to fix them, but until then, please try using the previous release at https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-05-28/ (Desktop) or https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/ (Lite)

explorigin, 2021-11-10:
@makermelissa Since this issue is closed, is there a new issue to track the bullseye fix (or should this be re-opened)?

PR #157 (2021-02-26): ST7789 Driver fix
Fixes #138.
This PR includes the following changes:
 * The fb_st7789v driver has been updated to work with the latest kernel and has offsets that work on all 3 st7789v displays.
 * The 1.14" driver was removed. It wasn't working on the latest kernel.
 * The 1.14" overlay was updated to use the fb_st7789v driver instead.
 * The code to run the pinning script was removed
 * Code to detect and remove the pin was added to the script
 * PiTFT Script was updated to be able to more efficiently update the FBCP orientations in config.txt and in a more configurable way.
 * PiTFT Script [... cut]