adafruit/Raspberry-Pi-Installer-Scripts #139: Can't get framebuffer running for ST7789_240x240 TFT on Pimoroni Pirate Audio device

Sindbad6, 2021-01-11:
Hello,
I have a Pirate Audio working on a RPi 0W.

For my purposes, I want to use the display as framebuffer device. Therefore I installed the Adafruit software.

Unfortunately  I can’t get it working as hoped. So my question is, how I get it working? I got the info, that I probably have to change the pin settings in the software, because they are different to the Adafruit device. Can somebody give me a hint where and how I have to do it?

ladyada, 2021-01-11:
@Sindbad6 we answered your post here
https://forums.adafruit.com/viewtopic.php?f=50&t=173756

closed 2021-01-11 by ladyada

makermelissa, 2021-01-11:
According to the forum post, it hanging, which could be the same issue as #138. You may want to watch that issue.

Sindbad6, 2021-01-11:
I asked at the pimoroni forum as suggested from Adafruit. The answer was to check the Pin assignment. Because of this I wanted to know where this can be done. I hope this question is not to difficult or a secret.

According to #138 I'm afraid it is not the same issue. Today I tried to implement it with an up to date system including the kernel and it is still hanging during reboot. This shouldn't happen anymore. So I think it has to do with the pin assignment.

ladyada, 2021-01-12:
yeah the framebuffer overlays are currently hardcoded for our hardware
@gadgetoid can someone please help @sindbad6 by submitting an overlay

ladyada, 2021-01-12:
@Sindbad6 where is the post you made on @pimoroni forum?

Sindbad6, 2021-01-12:
[https://forums.pimoroni.com/t/framebuffer-support-for-pirate-audio/15821](url) That was my post. In between the thread became a new direction and I'm not sure if the problem is just another pin setting which can be solved by software. Maybe the wiring at all is a problem.

E-man-dev, 2023-05-17:
I too have been searching for a framebuffer driver for a Pirate Audio board. While the name is a bit confusing, https://github.com/juj/fbcp-ili9341 lists "Pirate Audio ST7789 HAT" as a supported device (see pirate_audio_st7789_hat.h in its code list). I haven't tried it yet, but I will let you know how it goes when I do. If you try it before I do, please let me know if it works!

E-man-dev, 2023-08-13:
Bad news: looks like juj/fbcp-ili9341 uses software that is no longer part of Raspberry Pi OS. The maker mentions it if you read far down enough in the README. Looks like you could use an old version of Raspbian and it would work, but I’m not willing to go to that length. Anyone out there with enough talent to salvage the work of juj/fbcp-ili9341?