adafruit/Raspberry-Pi-Installer-Scripts #145: Update URL for wget in converted_shell_scripts

figital, 2021-01-26:
on this page ... (in this file)

https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/converted_shell_scripts/adafruit-pitft.sh

change the wget URL to this

https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/converted_shell_scripts/adafruit-pitft.sh

(otherwise the wget command throws a 404)

makermelissa, 2021-01-26:
Hi, we changed over to using python scripts recently. Is there a reason you are still trying to use the old shell script?

figital, 2021-01-26:
oops sorry about that didn't notice the py script. will close. (and try the  python version)

just can't get rid of an spidev0.0 not found error :p (trying to run the miniPiTFT)

closed 2021-01-26 by figital