adafruit/Raspberry-Pi-Installer-Scripts #146: tft stats demo: /dev/spidev0.0 does not exist

figital, 2021-01-27:
Had used this awesome miniPiTFT 1-2 years ago with the stats.py demo.

Trying to use it again with a fresher version of "Raspi OS". (On a version 3B board which may be the problem)

The new python installer successfully added the correct flags to the end of my /boot/config.txt file.

Just running into this error and a brick wall ....

```
Traceback (most recent call last):
  File "./stats.py", line 23, in <module>
    spi = board.SPI()
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 218, in SPI
    return busio.SPI(SCLK, MOSI, MISO)
  File "/usr/local/lib/python3.7/dist-packages/busio.py", line 186, in __init__
    self._spi = _SPI(portId)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 22, in __init__
    self._spi = spi.SPI(device=(portid, 0))
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/spi.py", line 167, in __init__
    raise IOError("{} does not exist".format(device))
OSError: /dev/spidev0.0 does not exist
```

Thanks for any info .... thinking the newer RPi OS's aren't as compatible with older Version 3 boards? I can try from scratch on an RP4 for debugging purposes .... hoping to use a lower wattage board though eventually the zero-w (with the mini-tft).

Just starting to learn how these display pins work to debug but still a n00b.

@figital

figital, 2021-01-27:
oops forgot to add that when i run "ls /dev/spi*" i just get this ...

`/dev/spidev0.1`

(and not also /dev/spidev0.0)

makermelissa, 2021-01-27:
Hi, I didn't realize you were trying to run the stats. Unfortunately, you can't run the stats example while you have the kernel driver installed. If you run the installer (without any parameters) and choose uninstall, it should fix the issue.

figital, 2021-01-27:
Thank you Melissa ... I'll give that a try! Closing this since it just seems like something I don't fully understand (yet). Will check for other support avenues going forward .... didn't seem like the freenode channel was used these days.

closed 2021-01-27 by figital

makermelissa, 2021-01-27:
No worries. You can always post in the forums for general help.

figital, 2021-01-27:
you're right ... ran the py script again, chose uninstall ... now it works :)

THANKS !!

jkrszwsk, 2022-11-29:
Hi, i was wondering if you could write up exactly what i should write in terminal in this case? I am new to this kind of things and i am lost. Thanks!