adafruit/Raspberry-Pi-Installer-Scripts #147: i2samp.sh / MAX98357 doesn't work on raspios 2021.01.12

avilleret, 2021-01-28:
On a Raspberry Pi 3 B+ and the  raspios 2021.01.12 release, no sound is played with the `i2samp.sh` script.
The installation seems fine as `aplay -l`  says the device exists but no sound with `speaker-test`.

The same script on the same board with the same wiring but with https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-05-28/ works fine.

So I guess the 2021.01.12 update broke something.

makermelissa, 2021-01-28:
You might want to try this page and see if it helps: https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/kernel-module-troubleshooting. Even though the guide is for a display, it's a workaround that uses an older version of the kernel.

avilleret, 2021-01-29:
thanks for the hint.
But after trying kernel 1.20210108-1 one more time from a fresh RaspiOS lite download everything seems to work fine.
The error might be somewhere else...

closed 2021-01-29 by avilleret