adafruit/Raspberry-Pi-Installer-Scripts #149: ST7789V drivers failure

89jd, 2021-02-04:
I am trying to install the ST7789 drivers, and I get the following errors.

Can you help?

Thanks

```
PITFT /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c: In function ‘st7789vada_fb_dirty’:
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c:64:13: error: ‘struct mipi_dbi_dev’ has no member named ‘enabled’
  if (!dbidev->enabled)
             ^~
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c: At top level:
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c:137:14: error: ‘mipi_dbi_release’ undeclared here (not in a function); did you mean ‘spi_res_release’?
  .release  = mipi_dbi_release,
              ^~~~~~~~~~~~~~~~
              spi_res_release
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c:138:2: error: ‘DRM_GEM_CMA_VMAP_DRIVER_OPS’ undeclared here (not in a function); did you mean ‘DRM_GEM_CMA_DRIVER_OPS’?
  DRM_GEM_CMA_VMAP_DRIVER_OPS,
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  DRM_GEM_CMA_DRIVER_OPS
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c: In function ‘st7789vada_probe’:
/home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.c:276:8: error: implicit declaration of function ‘devm_drm_dev_init’; did you mean ‘devm_drm_dev_alloc’? [-Werror=implicit-function-declaration]
  ret = devm_drm_dev_init(dev, drm, &st7789vada_driver);
        ^~~~~~~~~~~~~~~~~
        devm_drm_dev_alloc
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:279: /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module/st7789v_ada.o] Error 1
make: *** [Makefile:1808: /home/pi/Raspberry-Pi-Installer-Scripts/st7789_module] Error 2

PITFT Apt failed to compile ST7789V drivers!

```

89jd, 2021-02-05:
Ok, so I have managed to get it to build, I will add my changes later (pretty much just followed what it said to change it to) if everything else is ok.

I just get a blackscreen. The backlight is on.

I have tried _fbi_ / _cat /dev/random > /dev/fb1_, but both do nothing. Anywhere I can start to try and work out whats happening? I tried startx too and the Xorg logs look like they are working.

makermelissa, 2021-02-08:
I was able to reproduce this with the 1.14" Mini PiTFT and Raspberry Pi OS Lite.

nekuneko, 2021-02-08:
> 
> 
> I was able to reproduce this with the 1.14" Mini PiTFT and Raspberry Pi OS Lite.

Me too with a bare wired IPS 2.0" ST7789V display and Retropie Raspbian GNU/Linux 10 (buster) Linux retropie 5.10.14+ #1401 Mon Feb 8 14:19:47 GMT 2021 armv6l GNU/Linux. 😬

makermelissa, 2021-02-08:
It appears they made some code breaking changes in some kernel headers such as https://elixir.bootlin.com/linux/latest/source/include/drm/drm_mipi_dbi.h

nekuneko, 2021-02-08:
The kernel fix posted here https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/kernel-module-troubleshooting looks to fix the issue:

```
cd ~
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/rpi-pin-kernel-firmware.sh
sudo sh rpi-pin-kernel-firmware.sh 1.20201126-1
```

But it would be neccesary to modify the ST7789V driver in order to make it work natively with the new kernel and the new `drm_mipi_dbi.h` definitions.

makermelissa, 2021-02-08:
Ok, here's what I've figured out poking around. If you run the `rpi-pin-kernel-firmware.sh` script first, then it compiles and runs successfully. I was able to fix the items in the script where it compiled fine with kernel headers at 5.10 and after running the pin script, the screen displays something, but inverted. I've seen the inversion before on the 1.3" display and re-running the PiTFT script fixes it.  That consistency says my changes to the driver were good. However, with the changes to make it compile on the 5.10 headers, it will not compile with the 5.4 headers.

So what that means is that even though I can get it to compile successfully on the 5.10 headers, that will ultimately break with the fix for other issue (#138). So until we have #138 fixed, I really shouldn't mess with the driver unless I can add some conditional compilation so it works on both versions. For now, the plan is to modify the pitft installer to run the pinning script.

makermelissa, 2021-02-08:
I got it to compile conditionally, so the driver update is part of #151.

PR #151 (2021-02-08): Add pinning to pitft and fix driver
Ok, this PR does 2 things.
First it fixes the driver so it compiles under Kernel 5.10 and 5.4.
Second, it runs the pinning script as part of the PiTFT install process if the version is greater than 5.4.
Fixes #149.
closed 2021-02-08 by ladyada