adafruit/Raspberry-Pi-Installer-Scripts #155: Inverted Color on Braincraft ST7789

maditnerd, 2021-02-25:
Followed [BrainCraft Tutorial](https://learn.adafruit.com/adafruit-braincraft-hat-easy-machine-learning-for-raspberry-pi/display-kernel-module-install)

Color were inverted.

Fix it by changing these lines and running again
```sudo python3 adafruit-pitft.py --display=st7789_240x240 --rotation=90 --install-type=fbcp```
L42 [st7789v_ada.c](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/st7789_module/st7789v_ada.c#L42)
L32 [fb_st7789v.c](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/st7789_module/fb_st7789v.c#L37)

```
#define ST77XX_MADCTL_BGR 0x00
#define ST77XX_MADCTL_RGB 0x08
```

makermelissa, 2021-02-25:
Did you use an Adafruit BrainCraft HAT or other hardware?

maditnerd, 2021-02-25:
I used the Adafruit BrainCraft HAT

makermelissa, 2021-02-26:
Please try it again. I just finished rewriting the driver and updated the script. If it's still behaving incorrectly, there is supposedly a bgr parameter that I can look into enabling.

makermelissa, 2021-03-16:
Closing due to lack of response. We can always re-open. :)

closed 2021-03-16 by makermelissa