adafruit/Raspberry-Pi-Installer-Scripts #158: Static on lower 90% of screen after running adafruit-pitft.py 3.1.0

ashtronaut, 2021-02-27:
After following the BrainCraft tutorial & running adafruit-pitft.py, I get a screen with static as mentioned in the guide.  Afrer reverting the kernel back to 1.20201126-1. I get inverted colors as mentioned in the guide. I then run adafruit-pitft.py again as mentioned (version 3.1.0), but it hangs up while upgrading kernel. I left it to run for a few hours, but nothing changed. After aborting and cycling power the screen appears to work as expected which surprised me. After another power cycle, the screen shows static on the lower 90%.  The top 10% shows normally

makermelissa, 2021-02-27:
Did you let the script complete or did you stop it? 

3.1.0 should now work on the latest kernel. So it will unpin your kernel and depending on your device it may take awhile for it to complete the kernel update again. Like it takes 15 minutes or more on a Pi zero. The new driver won't compile until after the kernel upgrade is complete.

makermelissa, 2021-02-27:
You can manually just delete /etc/apt/preferences.d/99-adafruit-pin-kernel and run "sudo apt update" and "sudo apt upgrade" outside of the script, reboot, then run it again.

ashtronaut, 2021-02-28:
I think this did the trick.  I will continue to test & let you know if I come across any more issues.

ashtronaut, 2021-02-28:
This appears to work when i do soft reboots, but whenever i unplug the pi from power and plug it back in I get static on the
 screen again.

makermelissa, 2021-02-28:
Ok, I can take a look on Monday.

makermelissa, 2021-02-28:
I had a thought on this. I wonder if possibly some of the other software for the BrainCraft HAT is interferring with the SPI data. I have no problems on any of the Mini PiTFT and I can only recall seeing the issue on the BrainCraft HAT. Without audio or fan service installed, it appears to be working ok, but I'll monitor my Pi until Monday.

ashtronaut, 2021-02-28:
I tested your theory and I believe you are correct.  The fan service appears to be the culprit.  

I started with a fresh copy of pi OS & added the display, LED & servo libraries, and audio software with power cycles in between each.  After installing the fan service and performing a power cycle, I got static on my screen.

I will continue testing without the fan service & update if there are any further issues. Should I open a new issue for the fan service, or should we keep this thread updated?

makermelissa, 2021-02-28:
Excellent news because I was planning on rewriting it to make use of the built-in fan service anyways. #148 is already open in regards to the fan service, so you can add to that if you'd like.

ashtronaut, 2021-03-01:
I miss-spoke.  After re-testing, the static appears after I setup the audio.  In my original testing I may not have fully power cycled between installing audio & fan service.  When I power cycled after installing the fan service and the screen had static, I assumed the fan service was the culprit because it was the last thing I installed.  

I just went through the process again, twice, and the screen shows static after installing audio and power cycling.  When I skip the seeed-voicecard and install PiTFT & fan service, both the display and fans work as expected.  As soon as I introduce the software for seeed-voicecard & power cycle I get static on the screen.  

Furthermore, when I install the PiTFT software first and try to install seeed-voicecard, I get the following error.

makermelissa, 2021-03-01:
I reproduced this by installing the audio driver. hooked up a logic analyzer and on the SPI analyzer the error message I am getting is: `The initial (idle) state of the CLK line does not match the setting.`

Based on some searching it sounds like the SPI mode may be getting changed, but I will continue looking deeper into this.

makermelissa, 2021-03-01:
Looking at the source code of the audio driver (https://github.com/HinTak/seeed-voicecard/blob/v5.9/seeed-voicecard.c), I notice it messes with Interrupt requests and that actually seems like it might be more inline with the actual behavior than SPI mode being changed as changing the settings on the logic analyzer resulted in the same message no matter what mode I selected.

Also, unless I am actively moving the mouse cursor, there is no SPI data to be grabbed and that's the only time the screen redraws.

makermelissa, 2021-03-01:
I'm seeing lots of these messages in dmesg:
```
[17603.374512] fb_st7789v spi0.0: SPI transfer timed out
[17603.374578] spi_master spi0: failed to transfer one message from queue
[17603.374600] fb_st7789v spi0.0: fbtft_update_display: write_vmem failed to update display buffer
```
This occurs whether the seeed-voicecard service is running or not, yet while the problem remains active. What is more interesting is every time I restart the seeed-voicecard service I get the following set of messages:
```
[17646.924806] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
[17647.316419] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
[17647.604433] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
```

makermelissa, 2021-03-01:
I've been experimenting with turning overlays on and off. So far, I figured out that commenting out `dtparam=i2s=on` causes the problem to go away.

makermelissa, 2021-03-02:
So I tried enabling `dtparam=i2s=on` on a fresh copy of Raspberry Pi OS and the problem did not occur. Since this is a standard interface, I turned to other components that were installed. After some experimentation, I have narrowed it down to a line in `/etc/modules`, specifically the `snd-soc-wm8960` module. I'm not sure if this module is even required to use the sound, so if not, then just not loading it is a good solution for the time being. If it is, I'll try seeing if others have experienced bugs with it.

ashtronaut, 2021-03-02:
Good find!  So, the method of testing would be to start with a fresh copy of piOS, blacklist snd-soc-wm8960, install piTFT & seeed soundcard, and see if we get sound?

makermelissa, 2021-03-02:
Ok, this keeps getting stranger. First of all, the problem is actually intermittent, so apparently my results were not definitive. However, I noticed with all 3 parts commented out in the /etc/modules, the speaker still plays just as well as with the modules loading. However, the quality is subpar and sounds jittery (probably related to the display issues). So I'm going to keep seeing what can be removed until either the problem goes away or I can definitely narrow down the culprit.

makermelissa, 2021-03-02:
Ok, I may have found something useful. I looked at all of the repos that had been forked from the Hintak seeed-voicecard repo and found there was only a single one that had extra commits (https://github.com/M0Rf30/seeed-voicecard/commit/014b0b745b38910ba8c303cef1f0694717cf1574). In the commit, they remove the seeed-2mic-voicecard, which is what the audio portion of the BrainCraft appears to be modeled after, and it mentions "You can find WM8960 resources in latest RaspberryPi kernel sources".

Soooo, the issue may just be a driver conflict and it may be possible to get sound going without this installing anything from the repo at all. 🤞

makermelissa, 2021-03-02:
Ok, I tried installing the drivers from `https://github.com/waveshare/WM8960-Audio-HAT` and here are a few observations:
 * The driver works good on the BrainCraft HAT
 * Typing `speaker-test -c2` actually produces an output whereas the seeed one didn't
 * I was only able to test seeed sound output by playing YouTube videos
 * But, I couldn't get through the UI when the static issue was present, so didn't test sound at that time
 * I am still noticing the intermittent display noise issue with this driver as well
 * I notice the stutter with this driver as well when testing Youtube videos
 * However, for the speaker-test, now that I can play it. I noticed it only stutters when the noise issue is **not** present.

So it seems like we either have sound issues **or** display issues.

makermelissa, 2021-03-02:
Ok, I was able to navigate very slowly, but I can confirm that the YouTube videos also don't stutter when the display issue is present.

makermelissa, 2021-03-03:
Here's where this sits. I was never able to get it functioning on 5.10. There is either the display problem or the audio problem and I would consider this resolved if the display and audio issues were not present at the same time. I am going to put this on hold for the moment, but I do have at least a path to get people functioning.

I updated the PiTFT script to disable the code that removes the kernel pinning. The method that seems to be a workaround for now is:
* Run the kernel pinning script
* Run the PiTFT script (3.1.1) to install the display
* Run the install from v5.5 branch of seeed-voicecard

ashtronaut, 2021-03-03:
Thanks for your work on this Melissa.  I'll dig back into this within the next couple days & update this thread with any findings.

makermelissa, 2021-05-28:
So, I just installed with the latest pieces and it seems the problem has finally cleared up. I'm going to close for now, we can always re-open.

closed 2021-05-28 by makermelissa