adafruit/Raspberry-Pi-Installer-Scripts #159: Broken TFT Install Script FBCP

jgehrig, 2021-03-01:
Repro Steps:
```
python3 adafruit-pitft.py --display=28r --rotation=90 --install-type=fbcp
```

Error:
```
PITFT Adding FBCP support...
Installing cmake...
Downloading rpi-fbcp...
Uncompressing rpi-fbcp...
Building rpi-fbcp...
Installing rpi-fbcp...
Remove fbcp from /etc/rc.local, if it's there...
We have systemd, so install fbcp systemd unit...
Configuring boot/config.txt for forced HDMI
Traceback (most recent call last):
  File "adafruit-pitft.py", line 699, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "adafruit-pitft.py", line 684, in main
    if not install_fbcp():
  File "adafruit-pitft.py", line 431, in install_fbcp
    default_orientation = int(list(fbcp_rotations.keys())[list(fbcp_rotations.values()).index("0")])
UnboundLocalError: local variable 'fbcp_rotations' referenced before assignment
```

A git blame of the line in question (431) shows changes to this line were made recently: 290ea02e246664c0e324e79af1f08d2b2f1e6306.

@makermelissa Reverting your changes resolves the issues above. Can you take a look at this?

makermelissa, 2021-03-01:
Heh, classic mistake of uploading major changes on a Friday. #160 should fix it.

closed 2021-03-01 by makermelissa

jgehrig, 2021-03-01:
I've been there and done that before :)

Thanks for the quick fix!

PR #160 (2021-03-01): Fix script crashing on non st7789 displays
Fixes #159. Also removes warnings from device tree compiler.