adafruit/Raspberry-Pi-Installer-Scripts #166: support adafruit-pitft.py on Ubuntu 20.x+

craiglink, 2021-03-12:
The hardware and software work on Ubuntu 20.x ( verified on rpi4B, Ubuntu 20.10 (64bit) with st7789_240x240 ).  
The python install script just makes assumptions about the locations for the config.txt, cmdline.txt, and boot/overlay folder.
On Ubuntu 20.x these are in /boot/firmware/ and not just /boot/  
Once these locations were fixed the the displays worked in both console and fbcp mode. 

A hacked version of the original file is here -https://github.com/craiglink/Raspberry-Pi-Installer-Scripts/blob/master/adafruit-pitft-ubuntu.py, but some work could be given to make the original file support either Raspberry OS or Ubuntu

ladyada, 2021-03-12:
we don't support ubuntu (only raspbian is officially supported), but if someone submits a non-breaking PR we'd take a look at adding it!

makermelissa, 2022-06-01:
It seems adding the ability to pass some additional parameters and defaulting to current values would fix it without breaking anything.

PR #222 (2022-06-03): Added ability to specify boot directory as parameter
Fixes #166. I tested that it still works with defaults.
closed 2022-06-03 by makermelissa