adafruit/Raspberry-Pi-Installer-Scripts #171: Can't install i2s mic on Pizero

gavspav, 2021-03-30:
When running the install script I'm getting some errors:

make[1]: *** /lib/modules/4.19.71-rt24+/build: No such file or directory.  Stop.
make: *** [Makefile:38: clean] Error 2

make -C /lib/modules/4.19.71-rt24+/build M=/home/patch/Raspberry-Pi-Installer-Scripts/i2s_mic_module clean
make[1]: *** /lib/modules/4.19.71-rt24+/build: No such file or directory.  Stop.
make: *** [Makefile:35: all] Error 2

make -C /lib/modules/4.19.71-rt24+/build M=/home/patch/Raspberry-Pi-Installer-Scripts/i2s_mic_module modules
make: *** No rule to make target 'snd-i2smic-rpi.ko', needed by 'install'.  Stop.

makermelissa, 2021-03-31:
What version of Raspberry Pi OS are you running?

paulhobbel, 2021-04-15:
I am facing the same issue, running latest version of Raspberry OS

paulhobbel, 2021-04-15:
I solved it by running
```
sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel
```

makermelissa, 2021-04-15:
Great solution @paulhobbel and thanks for sharing what worked.

closed 2021-05-21 by makermelissa