adafruit/Raspberry-Pi-Installer-Scripts #175: PiTFT - Unable to turn on and off the display

aeromaxx, 2021-04-10:
I've just installed the "PiTFT 2.8" capacitive touch (240x320)" display using the latest 2.1.0 script, the screen is working.  But before I was able to turn the display on and off, using the commands below but now I am unable to as these paths don't exist.

#ON
sh -c '/bin/echo 0 > /sys/class/backlight/soc\:backlight/bl_power'
sh -c 'echo "1" > /sys/class/backlight/soc\:backlight/brightness'

#OFF
sh -c '/bin/echo 1 > /sys/class/backlight/soc\:backlight/bl_power'
sh -c 'echo "0" > /sys/class/backlight/soc\:backlight/brightness'

https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/converted_shell_scripts/adafruit-pitft.sh

makermelissa, 2021-05-21:
Hi, 2.1.0 is not the latest version of the script. We converted it to Python. Please try the 3.1.1 version. Here's how to run it: https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/easy-install-2

Thanks.

aeromaxx, 2021-05-29:
@makermelissa Thanks, but I have already tried that script and I don't get the option to turn the display on and off as and when I want, there is no mention of how to do that on the link you posted, I had to revert back to an older version of the script so I could turn the screen on and off when I want, the latest version doesn't have that functionality.

CrossVR, 2021-08-15:
I'm not sure why, but the ability to control the backlight was removed in commit 8bc1771bb2b1e40202058f39cc136620ffecfb39

makermelissa, 2021-08-16:
Yeah, it could be related.

makermelissa, 2021-09-09:
The ILI9341 driver is part of the Raspberry Pi OS and it looks like they took this out, so there's not much to do there. I was however able to control it using PWM and Python. Take a look at the Blinka on Raspberry Pi guide: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/pwm-outputs-servos#pwm-leds-3067131-4. I just changed the Pin to D18 and it faded the backlight in and out perfectly.

makermelissa, 2021-09-10:
Ok, I looked into this a bit more. `/sys/class/backlight` is only installed for the resistive touch screen because it is part of the STMPE driver. I have updated the guide page to include some alternative ways to control the backlight on the Capacitive screen.

closed 2021-09-10 by makermelissa

aeromaxx, 2021-09-13:
> The ILI9341 driver is part of the Raspberry Pi OS and it looks like they took this out, so there's not much to do there. I was however able to control it using PWM and Python. Take a look at the [...]

@makermelissa Thanks for all your work on this, I am still able to successfully turn the display on and off using the 0.8.0 version of this script on the latest version of raspbian, so I am confused as to why it's not possible to do the same with later versions of the script, even ones that have been converter over to Python.

makermelissa, 2021-09-13:
I'm not sure what you mean by the 0.8.0 version of the script. Could you point me to the one that works for you?

aeromaxx, 2021-09-13:
@makermelissa This one is working for me, and lets me turn on/off the display as in the original post at the start of this thread.

https://github.com/adafruit/Adafruit-PiTFT-Helper/blob/master/adafruit-pitft-helper

makermelissa, 2021-09-13:
Thanks, I didn't even know about that repo. I'll take a look. It's possible we diverged from there because it broke something else (perhaps a conflict with the resistive touch driver).