adafruit/Raspberry-Pi-Installer-Scripts #176: 2.8" PiTFT rotation not working

JamesRandom, 2021-05-08:
I am using the "easy install" script to set up the display:
```
sudo python3 adafruit-pitft.py --display=28r --rotation=270 --install-type=fbcp --reboot=yes
```

However, the display is the same whether I select `--rotation=90` or `--rotation=270`.

Hardware: Raspberry Pi 3B
OS:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian

JamesRandom, 2021-05-08:
BTW both the page with the instructions and the script are confusing with regard to rotation values. If you run the script interactively, it prompts for rotation values as:
```
Select rotation:
[1] 90 degrees (landscape)
[2] 180 degrees (portrait)
[3] 270 degrees (landscape)
[4] 0 degrees (portrait)
```

So 270 degrees is `3`.

But if you use the `--help` option, it says:
```
  --rotation INTEGER              Specify a rotation option (1-4) or degrees
                                  (0, 90, 180, 270)
```

So 270 degrees is `4`.

I didn't bother to try and work out which is correct, I just used `--rotation=270`!

makermelissa, 2021-05-17:
When you say "the display generated by X (no desktop)" and you saying you are running Raspberry Pi OS Lite?

I just tested on a Pi 4 and it was fine. I'll try on a Pi 3 next to see if that makes a difference (it shouldn't).

makermelissa, 2021-05-17:
Yeah, same behavior on the Pi 3. 
Here's 270:
[image]
Here's 90.
[image]

JamesRandom, 2021-05-18:
> When you say "the display generated by X (no desktop)" and you saying you are running Raspberry Pi OS Lite?

Yes, exactly. And with all software updated.

> Yeah, same behavior on the Pi 3.

OK. Presumably, user error. I'll try and work out what I am doing wrong. (I'll leave the issue open until I find out what is going on)

makermelissa, 2021-05-21:
Closing for now. If it's still an issue, we can always reopen :)

closed 2021-05-21 by makermelissa

JamesRandom, 2021-05-22:
I have (partly) found out what the problem is. I was starting my program (including running `startx`) from `/etc/rc.local`. 

If I run it from the command line, after the system has booted, then the orientation is as expected. Presumably, I need to run start and the program later in the bot sequence

JamesRandom, 2021-05-22:
No, that is not the problem. Now I am confused.

If I start X from command line:
```
sudo startx
```
The the orientation is correct (it stays as it is)

But if I run it as root:
```
sudo -E su
startx
```
Then the display flips and is the wrong way round 😕

JamesRandom, 2021-05-22:
I thought it might be something to do with the environment so:

```
export DISPLAY=:0.0 
export FRAMEBUFFER=/dev/fb1 
sudo startx
```
Works as expected

But with
```
sudo DISPLAY=:0.0 FRAMEBUFFER=/dev/fb1 /usr/bin/startx
```
And
```
export DISPLAY=:0.0 
export FRAMEBUFFER=/dev/fb1 
sudo -E /usr/bin/startx
```
It does not work: the display is flipped back to the default setting

JamesRandom, 2021-05-22:
It looks like I don't need to set FRAMEBUFFER, and that may be what is causing the problems. I must have got that from an older version of the setup instructions. 😞