adafruit/Raspberry-Pi-Installer-Scripts #179: Touch rotation issues on PiTFT 3.5

pietvanzoen, 2021-05-21:
Hey folks.

I'm using a PiTFT 3.5 in landscape and cannot get the touch screen to rotate correctly. The touch screen seems to be constantly set to portrait mode. I've tried a few different things including `lcd_rotate=3` and what's [suggested here](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues/59#issuecomment-569577906) with no luck.

Any tips would be appreciated.

Here's the relavant config in `/boot/config.txt`:
```conf
hdmi_group=2
hdmi_mode=87
hdmi_cvt=800 533 60 1 0 0 0
display_hdmi_rotate=2

# --- added by adafruit-pitft-helper Fri May 21 11:17:28 2021 ---
hdmi_force_hotplug=1  # required for cases when HDMI is not plugged in!
dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=pitft35-resistive,rotate=90,speed=20000000,fps=20
# --- end adafruit-pitft-helper Fri May 21 11:17:28 2021 ---
```

I'm using this with https://github.com/Z-Bolt/OctoScreen.

makermelissa, 2021-05-21:
Hi, I have a few questions about your setup.

Which board are you using this with? It appears to be Raspberry Pi, but I want to verify.
If RasPi, are you using Raspberry Pi OS? It appears to be Raspberry Pi OS, but I want to verify.
If RasPi OS, are you using lite or desktop? I'm guessing Desktop, but want to be sure.
What was the install method that you used? Probably doesn't matter, but it's helpful in reproducing the issue.
Did you install as FBCP or console? I'm fairly sure you used the FBCP method because of display_hdmi_rotate.

Finally, did you try changing the value of display_hdmi_rotate= to something else?

Thanks.

makermelissa, 2021-09-09:
Closing due to lack of response and not enough information to reproduce the issue.

closed 2021-09-09 by makermelissa

AlexKlimaj, 2022-08-02:
I just ran into this same problem on the 2.8" capacitive touchscreen and a Pi 4. I had to change config.txt to get the touchscreen to match the display. I used the setup script and it produced the following.

`dtoverlay=pitft28-capacitive,rotate=90,touch-swapxy=true,touch-invx=true`
changed to
`dtoverlay=pitft28-capacitive,rotate=90,touch-swapxy=false,touch-invx=false`