adafruit/Raspberry-Pi-Installer-Scripts #183: Unable to run this on db410c

RajithShetty619, 2021-07-05:
running the script gives me the following error

`
Installing build requirements - this may take a few minutes!

Hit:1 http://deb.debian.org/debian sid InRelease
Ign:2 http://obs.linaro.org/linaro-overlay-sid/sid ./ InRelease
Hit:3 http://obs.linaro.org/linaro-overlay-sid/sid ./ Release
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package swig3.0
E: Couldn't find any package by glob 'swig3.0'
E: Couldn't find any package by regex 'swig3.0'
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package raspberrypi-kernel-headers
Cloning libgpiod repository to /tmp/libgpiod.sMKx

Cloning into '.'...
remote: Enumerating objects: 29, done.
remote: Counting objects: 100% (29/29), done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 6247 (delta 16), reused 14 (delta 14), pack-reused 6218
Receiving objects: 100% (6247/6247), 1.07 MiB | 202.00 KiB/s, done.
Resolving deltas: 100% (4310/4310), done.
Building libgpiod

autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: configure.ac: creating directory autostuff
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
configure.ac:161: error: possibly undefined macro: AC_LIBTOOL_CXX
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:163: error: Unexpanded AX_ macro found. Please install GNU autoconf-archive.
configure.ac:177: error: Unexpanded AX_ macro found. Please install GNU autoconf-archive.
autoreconf: /usr/bin/autoconf failed with exit status: 1
cp: cannot stat 'bindings/python/.libs/gpiod.so': No such file or directory
cp: cannot stat 'bindings/python/.libs/gpiod.la': No such file or directory
cp: cannot stat 'bindings/python/.libs/gpiod.a': No such file or directory
`

makermelissa, 2021-07-07:
Try running it with the `-l` or `--legacy` flag.

RajithShetty619, 2021-07-08:
unfortunately didnt work still getting the same cp errors

makermelissa, 2021-09-09:
There may be a gpiod package available for your board that can be installed with `apt` or `apt-get`.

makermelissa, 2021-09-13:
It's installable via `apt`. I have updated the guide for that board.

closed 2021-09-13 by makermelissa